balena-io/etcher #1930: Feature Suggestion - Debian based distros

linuxlite, 2017-12-19:
Add the following to the postinst file to automate repo and key installation:

```
echo "deb https://dl.bintray.com/resin-io/debian stable etcher" | sudo tee /etc/apt/sources.list.d/etcher.list

sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
```

Add the following to the postrm file to automate repo and key removal:

```
sudo rm /etc/apt/sources.list.d/etcher.list

sudo apt-key del 379CE192D401AB61
```

lurch, 2017-12-19:
I guess that works if you're downloading the `.deb` file and installing it directly with `dpkg`, but the instructions we provide [here](https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64) are designed so that you don't download the `.deb` directly, but instead install it with `apt-get`.

AFAIK if we did what you suggest above, and then a user followed the instructions we provide, they'd end up with duplicate entries in their `/etc/apt/sources.list.d/etcher.list` ? Is it possible to make it conditional so that it doesn't update `/etc/apt/sources.list.d/etcher.list` if it already exists?
And I guess there's the 'subtle wrinkle' that if a user installs a snapshot `.deb` file rather than a production `.deb` file, does that mean we'd need to add the snapshot repo to `/etc/apt/sources.list.d/etcher.list` instead of the production repo? (see #1767 for more of an overview)

And I've no idea what standard Debian policy is regarding `postrm` steps (ping @dlech ) but I know I'd be pretty annoyed if I manually created `/etc/apt/sources.list.d/etcher.list` and then uninstalling `etcher-electron` automatically deleted it?

As a separate question, is it worth adding that `apt-key del` command to [here](https://github.com/resin-io/etcher#uninstall) ?

linuxlite, 2017-12-19:
@lurch I'll write both of the scripts for you to test under all possible scenarios.

linuxlite, 2017-12-19:
Ok this is fairly simple in Ubuntu.
I've tested this multiple times, works like a charm.

Explanation - you can't add duplicates of the same key so there is no issue there. I've put a check in to see if an existing /etc/apt/sources.list.d/etcher.list file exists, if it does, do nothing, if it doesn't, add it:

Modified **postinst**:

```
#!/bin/bash

# Link to the binary
ln -sf '/opt/Etcher/etcher-electron' '/usr/local/bin/etcher-electron'

# Add the Etcher key
apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61 > /dev/null 2>&1

# Check if the /etc/apt/sources.list.d/etcher.list file exists
if [ -f "/etc/apt/sources.list.d/etcher.list" ]; then exit; else echo "deb https://dl.bintray.com/resin-io/debian stable etcher" | sudo tee /etc/apt/sources.list.d/etcher.list :; fi

```

Modified **postrm** (best practice is to clean up, don't leave anything behind if you're not going to use it):

```
#!/bin/bash

# Delete the link to the binary
rm -f '/usr/local/bin/etcher-electron'

# Remove the Etcher repo
rm /etc/apt/sources.list.d/etcher.list

# Remove the Etcher key
apt-key del 379CE192D401AB61 > /dev/null 2>&1
```
If you are really worried about Debian policy, look at the current Lintain output in the existing .deb files Etcher is currently serving.

This now negates the need for any 'instructions' other than download the deb, install, drink beer.

Please give this serious consideration. It would make everyones lives including Etchers, so much easier.

Cheers :)

dlech, 2017-12-19:
I don't understand why you would set up the remote repository *after* installing the package. The point of setting up the repository is so that you can install the package with `apt`. Also having a package that removes an apt source when it is uninstalled is unprecedented, so I suspect it would catch a lot of people off guard.

> As a separate question, is it worth adding that apt-key del command to here ?

Actually, I wouldn't use `apt-key` at all. The "right" way (according to some Debian maintainers I talked to) is to just put the key in `/etc/apt/trusted.gpg.d`. (`apt-key` is no longer a required package in Debian since Stretch, so as time goes on, I expect fewer people to have it installed.) So, in the README, I would replace

    sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61

with

    curl https://bintray.com/user/downloadSubjectPublicKey?username=bintray | sudo gpg --dearmor --output /etc/apt/trusted.gpg.d/bintray.gpg

This way, if people want to inspect the key first instead of piping directly to gpg, they can.

    curl https://bintray.com/user/downloadSubjectPublicKey?username=bintray -O -J
    # inspect bintray-public.key.asc
    sudo gpg --dearmor --output /etc/apt/trusted.gpg.d/bintray.gpg bintray-public.key.asc
    rm bintray-public.key.asc

Then you just have to add this line to the unintall steps:

    sudo rm /etc/apt/trusted.gpg.d/bintray.gpg

KlfJoat, 2018-01-26:
FYI, one might set up the repository after installing because that install flow is simpler and requires no CLI.  (download .deb, install)  There are a few third party packages I use that install themselves in this way.  

As long as you make the postinst script idempotent, what's the harm?

I disagree with the postrm script, though.  Just bc I remove your package doesn't mean I want to remove your repository.  Maybe I'm troubleshooting or testing something.

erik-alatest, 2018-02-13:
As per the latest instructions:
```bash
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.ed8DcRlOKk/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data
```
What to do now?

dlech, 2018-02-13:
> What to do now?

See https://github.com/resin-io/etcher/issues/1930#issuecomment-352845004

lurch, 2018-02-22:
We've had a few sporadic reports of the pgp.mit.edu keyserver being temporarily unavailable, but luckily it usually seems to be a fairly short outage.

closed 2018-10-15 by jviotti