balena-io/etcher #1935: cmd prompt

nitestryker, 2017-12-22:
- **Etcher version:**  V1.2.1 
- **Operating system and architecture:** Windows 10 x64
- **Image flashed:** No
- **Do you see any meaningful error information in the DevTools?**

Humor me on a theory.  I am using windows 10 x64 and I have issues with other programs running cmd  prompts in the past with out administrative permissions .  I've actually had to run command prompt  with Administrative permissions to get certain programs to run correctly

now here is my point.  I have had no luck getting my image to burn  I noticed that it runs a cmd prompt and my theory is maybe the cmd prompt needs to be run  with administrative privileges  for it to work. because after i get the UAC screen asking if its ok to run the cmd prompt and i say yes it just doesn't go any further then that.  I was trying to figure out a  "work-around" for this but have  had no luck to test my theory. 

let me know your thoughts 

i was wondering if testing by "hard coding" it  to run the cmd prompt as administrator if that might fix the problem.

benp13, 2018-01-03:
I had the same problem. I restarted Etcher using "run as administrator" and it works ok now.

lurch, 2018-01-03:
> i was wondering if testing by "hard coding" it to run the cmd prompt as administrator if that might fix the problem.

I'm no Windows expert, but I believe that's exactly what the "UAC screen" is asking. We use https://github.com/jorangreef/sudo-prompt to get the administrative privileges that Etcher needs to be able to perform low-level disk writes.

jhermsmeier, 2018-01-05:
The writer process / the cmd is elevated on Windows (hence the UAC prompt), but there were some issues we fixed recently, which might be related: #1950 and #1910 – those will be released in the next version shortly.

PS: @lurch on Windows we're using a native binding that @jviotti wrote, not sudo-prompt, because there were issues with that

lurch, 2018-01-08:
Ahh, I thought we were using sudo-prompt because it's still in `package.json` ? https://github.com/resin-io/etcher/blob/master/package.json#L93

jviotti, 2018-01-08:
We use `sudo-prompt` for GNU/Linux and macOS, but our custom native
module on Windows.

closed 2018-10-15 by jviotti

captainmustard, 2024-02-11:
Not sure if this comment will post since this issue is closed but...

I had this issue and the problem was that `C:\Windows\system32` was missing from the PATH on my windows install. Was a quick easy fix. Had to close and restart etcher for it to begin working.

hope this helps someone.

lurch, 2024-02-11:
@captainmustard See also #3621