balena-io/etcher #1936: Add supported file types to macOS Info.plist

rhwood, 2017-12-23:
- **Etcher version: 1.2.1**
- **Operating system and architecture: macOS 10.13.2**
- **Image flashed: resin.img from https://files.resin.io/resinos/raspberry-pi2/2.2.0%2Brev1.dev/image/resin.img.zip**
- **Do you see any meaningful error information in the DevTools? No**

It would be helpful if this app could include supported image types in the [Document Types](https://developer.apple.com/library/content/documentation/General/Reference/InfoPlistKeyReference/Articles/CoreFoundationKeys.html#//apple_ref/doc/uid/TP40009249-101685-TPXREF107) in the Info.plist file for macOS. This would make Etcher appear in the "Open With..." menu item in the File menu or context menu for supported image types in the Finder.

lurch, 2018-02-28:
This might also require #1386 to be implemented first?

rhwood, 2018-02-28:
Its likely that both this and #1386 be implemented at the same time for #1386 to work on macOS.