balena-io/etcher #1942: Etcher bad experience: it destroyed my MacBook charger and is crashing my Windows PC.

Silver978, 2018-01-01:
- **Etcher version: 1.2.1**
- **Operating system and architecture: macOS High Sierra and Windows 10**
- **Image flashed: Raspbian Stretch With Desktop**
- **Do you see any meaningful error information in the DevTools? No**

I used Etcher many times to flash a single SD Card but sometimes the operation was done correctly and other times no.
I had a big problem with this program.
Firstly, I used it on my MacBook Pro with a macOS High Sierra but it corrupted my SD Card and it also destroyed my MacBook charger: for a week I couldn't use my laptop.
So, I used Etcher on a computer with Windows 10 but every time I tried flashing my SD Card my computer crashed and the main Hard disk was overloaded with operations: the only thing to do was reset the computer.
After this, I came up with the idea that Etcher used too much computer's resources and in this way the charger of my MacBook blowed up.
Same for my PC, Etcher was using all the resources.
This program has a lot of potential and is very easy to use, but I had only bad experiences with it.
Hope this can help with your development!

jviotti, 2018-01-02:
Hi @Silver978 ,

> I used Etcher many times to flash a single SD Card but sometimes the operation was done correctly and other times no.

Etcher makes sure every single byte was written correctly, otherwise it will complain with a validation error, usually caused by faulty SD Cards/readers, etc

> Firstly, I used it on my MacBook Pro with a macOS High Sierra but it corrupted my SD Card and it also destroyed my MacBook charger: for a week I couldn't use my laptop.

Etcher won't corrupt your SD Card. The worst it can happen is that the image you flash to contains a partition table with file-systems not recognized by macOS. You can wipe the partition table and re-format it to use it for data storage again.

Regarding the MacBook charger, there is no way an application running in user mode can mess with something like a charger. The charger simply puts energy into your laptop, and won't be affected by anything running on your computer. 

> After this, I came up with the idea that Etcher used too much computer's resources and in this way the charger of my MacBook blowed up.

Computer resource usage is completely unrelated to a charger's failure, that simply drives energy into your laptop's battery.

> So, I used Etcher on a computer with Windows 10 but every time I tried flashing my SD Card my computer crashed and the main Hard disk was overloaded with operations: the only thing to do was reset the computer.

I don't know what your specs are, but the resources that Etcher uses should be absolutely fine on any hardware that is not very old. Regarding I/O operations, Etcher reads the image from the hard drive and writes the bytes to an external drive. It won't produce more I/O operations that copying a file from your hard drive to your external drive.

What you say "resources", are you talking about CPU usage, RAM, or what precisely?

Silver978, 2018-01-02:
Thank you for all the information and your time, my doubts were cleared.
Unfortunately I can't test if it's the CPU or RAM because my PC is completely freezed and the hard disk usage is at 100%.
Probably I have some problems with my hardware.

jhermsmeier, 2018-01-05:
Closing, as this is not a bug at this stage; please reopen if necessary.

closed 2018-01-05 by jhermsmeier