balena-io/etcher #1944: Etcher running in the background CPU at 92%  (Battery broke too, maybe related)

wwwhtml, 2018-01-03:
A few days ago Etcher was downloaded and used once. After that time I quit the application. 

Today it was deleted when it was noticed it was running in the background with a CPU load of about 92%. 

Etcher still the Recycle Bin if any of its files are of your interest.

[image]

- **Etcher version: 1.2.1**
- **Operating system and architecture: MacOS Sierra 10.12.6**
- **Image flashed: Rasbpian**
- **Do you see any meaningful error information in the DevTools?**

jhermsmeier, 2018-01-03:
We've had instances of this before; but this should be gone once https://github.com/resin-io/etcher/pull/1945 is merged

jhermsmeier, 2018-01-03:
Duplicate of #1288

closed 2018-01-03 by jhermsmeier

wwwhtml, 2018-01-04:
Thank you folks!!