balena-io/etcher #1951: WHY DOES ETCHER NEED PERMISSION???

hibruh808, 2018-01-04:
Hi, I just need to know why etcher needs permission to access the system. My mom won't let me download the file if I can't clarify what it does.

jhermsmeier, 2018-01-04:
Hi @hibruh808, Etcher needs administrative privileges only to write to the storage device (an SD-card, hard drive, etc.), because access to those is protected by the operating system, as your operating system uses them, and so ensures that normal programs can't interfere and cause data-loss or other issues. When rewriting the contents of a storage device, the program (in this case Etcher) needs to ask the operating system for permission, which in turn asks the user for permission. And that's why Etcher asks for those privileges to be granted before flashing an SD-card, for example.

jhermsmeier, 2018-01-04:
If something's unclear, or you'd like to know more, feel free to ask!

hibruh808, 2018-01-04:
Ty so much

closed 2018-01-04 by jhermsmeier