balena-io/etcher #1955: Error: stdout maxBuffer exceeded

jhermsmeier, 2018-01-05:
- **Etcher version:** 1.3.0
- **Operating system and architecture:** Linux
- **Error information:** https://sentry.io/resinio-57/etcher-v130/issues/429230586/

bdfoster, 2018-01-06:
Reproduced on Ubuntu 16.04.2

potatowave, 2018-01-06:
Reproduced on Ubuntu 16.04.3

Seems to happen at 73% copying a 1.16GB ISO onto a 16GB USB stick.

nicklassundin, 2018-01-06:
It seems to happen randomly within spectrum of 30 - 70% into the process
4.9 GB .img to 16 GB Micro-SDHC
Ubuntu 16.04 64-bit

Cyb3rpunk-76, 2018-01-06:
Reproduced on Mint 18.2 Sonya, always on 65%, 32 Gb Micro-SDHC.

Timodz, 2018-01-06:
Reproduced on Mint 18.2 Sonya

krisYanachkov, 2018-01-06:
Etcher version: 1.3.0
Operating System: Mint 18.1 64-bit
SDHC: 8GB

SLiX69, 2018-01-06:
Get this too on 17.10 with Etcher 1.3

JGuillaumin, 2018-01-06:
Same error while flashing Raspian on a SD card, from Ubuntu 16.04
Finally I used `dd` command.

closed 2018-01-08 by jhermsmeier

Lewiscowles1986, 2018-01-18:
Ooh

[image]

Looks like this is still the case in xenial repos. I get it, there are 4 other issues pending fix

[image]
[image]

lurch, 2018-01-18:
We updated the bintray repos so that 1.2.1 is the latest version https://bintray.com/resin-io/debian/etcher
but I guess apt / dpkg must not downgrade packages if a newer version of the package is already installed :-/

We're hoping to release Etcher 1.3.1 soon, but as a temporary workaround you can do:
```
sudo apt-get remove etcher-electron
sudo apt-get update
sudo apt-get install etcher-electron
```
to manually downgrade to 1.2.1

manuti, 2018-01-25:
I know the issue is closed and I don't want to complaint your software but I think this bug corrupt two of my MicroSD. The way it was broken doesn't allow me to use again. The MicroSD appears to be locked.

lurch, 2018-01-25:
@manuti Does https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives help? If not, could you provide more details about what "appears to be locked" means please?  :)
(and BTW Etcher v1.3.1 has now been released, so please try upgrading to that :grinning: )

manuti, 2018-01-25:
I've been using Etcher for a long time, for sure before the 1.0 release. A couple of weeks ago I start preparing an *armbian* image on a Samsung MicroSD card and I end with a mess of nested partitions and unrecoverable state. I bougth a new Lexar MicroSD after one try I end with errors and I can't format or change the partitions.
I try your recipes about recovering broken drives:

* On Windows: I can perform all the steps, but the MicroSD is not changed
* On Mac: I can't and the ends with `Error: -69877: Couldn't open device`
* On Linux: ...

lurch, 2018-01-25:
> I end with a mess of nested partitions and unrecoverable state.

That sounds like https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos ? If you provide a link to the particular image you're using, we'll try investigating this ourselves.

manuti, 2018-01-25:
The images are the community armbian version for Amlogic 9XX, you can find in this link: https://forum.armbian.com/topic/2419-armbian-for-amlogic-s905-and-s905x/
I share also some images related to the issue. Thanks for you patience.
[image]
[image]
[image]

lurch, 2018-01-26:
Oh, is this bit what you meant by "a mess of nested partitions" ?
[image]

That's perfectly normal, and is a "feature" of MBR partitioning schemes. Basically an MBR can only store 4 primary partition definitions, so to get around that limitation you can create one of the primary partitions as an extended partition (in this case partition 2), and then create multiple logical partitions (in this case partitions 5 and 6) inside that extended partition. See these links for more of the gory details: https://en.wikipedia.org/wiki/Master_boot_record  https://en.wikipedia.org/wiki/Extended_boot_record
Under Linux the primary partitions are always labelled 1 to 4, and the extended partitions are always labelled starting at 5 (regardless of how many primary partitions there are).

Looking at the forum thread you linked to in #1960 , it seems like you're also having problems with other disk-flashing tools, and the problem you're having (whatever its underlying cause) isn't just limited to Etcher?

manuti, 2018-01-26:
Thanks for the info about partition.
About my comment in the other forum, I only use Etcher for write images to use with my boards. After having problems with Etcher I decide to try a different approach, but I think the damage is done. My microSD cards are broken and I started having problems with all the other image writer software. In the end I decided to manual partition the microSD, mount the images and copy the BOOT and ROOTFS. In this way the card works and I can my board.
Today I'm going to recover with Ease Windows software if I can't recover, if you wishes I can send my two microSD card for further investigation on your side.
I'm not complaining, I'm only commenting because I thinks this can help someone. 
Thanks again for your patience, your time and your software.

jhermsmeier, 2018-01-26:
> if you wishes I can send my two microSD card for further investigation on your side.

@manuti thank you for offering that – it would indeed be very helpful in figuring out what's going on there! Let us know how it goes & should it fail, we'll figure out the shipping!

lurch, 2018-01-26:
Unless your microSD cards are truly flaky at the hardware level, I think it's actually impossible to 'break' them from software.
In the worst possible case (if https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives doesn't help), you should be able to use `sudo dd if=/dev/zero of=/dev/xxx bs=4M` on Linux which will overwrite your entire SD card with zeros (i.e. blank it), which may take some time, and then you should be able to use any disk-partitioning (or image-writing) tools as normal.
I don't have any Amlogic boards so I obviously can't help investigate why your SD cards weren't booting properly when flashed with an image but then did when you partitioned them manually.

manuti, 2018-01-29:
@jhermsmeier Finally I can confirm that I have two 16GB microSD "bricked" 
How and where I can send you the cards?

jhermsmeier, 2018-01-29:
@manuti can you shoot us an email at hello@etcher.io and we'll sort it there?

commit 96dddecd (2018-01-08): fix(writer): Silence IPC output on stdout (#1961)

This was causing the stdout maxBuffer size to be exceeded
when flashing larger images (or having flashes that took a while).

Fixes #1955 

Change-Type: patch
Changlog Entry: Fix "stdout maxBuffer exceeded" error on Linux