balena-io/etcher #1956: error on install

ben-h253, 2018-01-05:
- **Etcher version:**current
[image]

- **Operating system and architecture:**Windows 8
- **Image flashed:**?
- **Do you see any meaningful error information in the DevTools?**n.a.

vtabary, 2018-01-06:
Same here with Windows 10.
So I'll use the 1.2.1 for now : https://github.com/resin-io/etcher/releases/

jviotti, 2018-01-10:
Hi @ben-h253 and @vtabary ,

Were you running the 64-bit or 32-bit versions? Also, were you using the installer or the portable versions? Do you have any anti-virus installed?

vtabary, 2018-01-10:
Hi @jviotti ,

I'm running the 64-bits version of winfodows 10. And yes, I use Avast antivirus.

jhermsmeier, 2018-01-11:
We've discovered that this is caused by the lack of certain C runtime libraries on some Windows systems that the WinUSB driver generator requires to function.

commit 45c72f0c (2018-01-18): fix(usb): Ignore errors if winusb doesn't load (#1970)

Due to some Windows systems missing certain C runtime libraries
(Visual C/C++ 2012 / 2015 Redistributables), we ignore errors when loading
this module until we can ensure distribution of those along with it.

Change-Type: patch
Changelog-Entry: Fix "The specified module could not be found" on Windows
Fixes #1956
closed 2018-01-18 by jhermsmeier