balena-io/etcher #1957: Etcher Pro advertisements

combs, 2018-01-07:
- **Etcher version:**
1.1.2 

- **Operating system and architecture:**
OS X

- **Image flashed:**
N/A

- **Do you see any meaningful error information in the DevTools?**
N/A

Hello! I've enjoyed using Etcher. Thank you for making it. 

I was surprised to see an Etcher Pro advertisement after successfully transferring an image. 

The Etcher Pro site says,

> Etcher Pro is the next stage in Etcher’s journey, and an expansion of Etcher’s vision and focus
Etcher is already the best and fastest way for writing to disk images, in fact it’s currently writing 500,000 SD Cards & USB Drives per month! Now, we are working on a way for you to write to more devices, with the same ease of use and streamlined interface you have come to expect from Etcher, and completely independent from your computer.
> 
> Etcher Pro is a stand-alone hardware device that allows you to write to multiple cards or usb disks at once, at extreme speeds. Compared to a traditional Disk Duplicator, Etcher Pro is faster and less expensive, while at the same time easier to use and packed with features, so that you can do much more than just copy SD Cards.
> 
> Etcher Pro is a device that copies a lot of SD cards very quickly, right? So the same functionality, but faster.

Does this create a conflict of interest for maintaining Etcher, the software product? If you sell a hardware device, it seems that you have less incentive to improve the equivalent software. 

I know that resin.io has put a lot of effort into creating this software. And to be clear, it doesn't seem to me like you harbor any ill intent whatsoever. But the cookie jar's lid has been taken off.

I don't see an easy solution—perhaps sending Etcher-the-software off into the community to maintain? Would the Pi Foundation be interested, since it relates so directly to their core product?

Forgive me for addressing this so directly. 

Best wishes,
Chris

alexandrosm, 2018-01-07:
Hi Chris,

If EtcherPro succeeds, it means that we will have more funds to invest into
Etcher, since both will run substantially the same software. EtcherPro will
in effect be Etcher, in hand-picked hardware. Everyone will still be free
to use Etcher, whereas EtcherPro will address the professionals, for whom
it's valuable to have the perfect hardware for the job. Notice that we're
not converting Etcher to paid software, or anything like that. I will make
sure to clarify that in the EtcherPro page. It is in EtcherPro's favour to
have a huge installed base for Etcher, and it is in Etcher's favour to have
a guaranteed revenue stream.

At the same time, since Etcher is open source, anyone is free to fork,
improve, and release their own variant. I hope I've made clear, however,
that Etcher isn't going anywhere :)

lurch, 2018-01-10:
I believe this issue has been resolved, so closing it.

closed 2018-01-10 by lurch