balena-io/etcher #1958: crash

fcailloux, 2018-01-07:
1.30
peppermint os
any image crash
buffer error

- **Etcher version:**
- **Operating system and architecture:**
- **Image flashed:**
- **Do you see any meaningful error information in the DevTools?**

alexandrosm, 2018-01-07:
Hi there, thanks for the report. We've rolled back the 1.3.0 release,
please re-download from the website until we can fix this issue.

jhermsmeier, 2018-01-08:
@fcailloux can you give us more detail on the "buffer error"? Might this be a duplicate of #1955?

lurch, 2018-02-28:
Closing, as assuming this was fixed by the Etcher 1.3.1 release.
@fcailloux Please feel free to leave more details if you're still having problems with the latest version.

jhermsmeier, 2018-02-28:
Duplicate of https://github.com/resin-io/etcher/issues/1955

closed 2018-02-28 by jhermsmeier

lurch, 2018-02-28:
Whoops, guess I forgot to actually click the "Close" button :rofl: