balena-io/etcher #1960: Error "You're not authorized to perform this operation"

evanoloughlin, 2018-01-08:
I get the following error when trying to write an image to my SD card:
> You're not authorized to perform this operation
> Please ensure you have necessary permissions for this task
This occurs after ~ 86% has been completede
Etcher is grand SU permissions when the flash begins

- Etcher version: 1.3.0
- Operating system and architecture: Ubuntu 16.04.3 (Linux 4.4.0-104-generic #127-Ubuntu SMP Mon Dec 11 12:16:42 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux)
- SD card image created using DD

lkmcdo, 2018-01-09:
Same feedback but using the appimage on OpenSUSE 42.3 (up to date patches on x86_64, idk where to find a full string).

Granted sudo when prompted but did not run etcher using SU.

manuti, 2018-01-25:
I think this can be related to a broken SD after using Etcher 1.3.0 version @
Duplicate of #1955

lurch, 2018-01-26:
@manuti Nope, this isn't a duplicate of that issue ;-)  Even when an SD card is 'broken', Etcher works at such a low level that it's still able to write a new image to that SD card.

@evanoloughlin It seems weird for Etcher to suddenly have a permissions problem 86% of the way through writing.
What SD card are you using (make, model, size, etc.) ? What SD card-reader are you using? How big is the image that you're trying to write to your SD card?

Oh, and I've just noticed that you said "Etcher version: 1.3.0" - that's a known buggy version, please upgrade to Etcher v1.3.1 and see if you still have problems.

manuti, 2018-01-26:
@lurch OK, maybe is not duplicated but I think is related to the other issue, you can see my post in armbian forum https://forum.armbian.com/topic/2419-armbian-for-amlogic-s905-and-s905x/?do=findComment&comment=47039 
I have the same ~80% error in a 16GB Samsung Evo used with Recalbox without problems and a completely new 16GB Lexar 300x. I try  with 3 different SD adaptor, 2 different USB to microSD reader ... always thinking that's a problem with the images.

gehaslam, 2018-02-20:
I had the same problem using both Win32 disk imager and and etcher (WINDOWS 7  OS). Wes successful with etcher after System reboot. It's rare that I've experienced these kinds of problems using Linux.

lurch, 2018-02-22:
Thanks @gehaslam  Good to hear that the "try turning it off and on again" still fixes Windows problems :rofl:

lurch, 2018-02-28:
@evanoloughlin Do you still have problems with Etcher 1.3.1 ?

evanoloughlin, 2018-03-01:
@lurch - I have not reproduced the issue so far with the latest version of Etcher (1.3.1)

jhermsmeier, 2018-03-01:
@evanoloughlin that's good news :)
Closing this then, as we've also not seen this come up in quite a while

closed 2018-03-01 by jhermsmeier

raydot, 2018-03-26:
<raises hand> I've got the issue.  Tried from both Etcher and Etcher CLI on OSX Sierra 10.12.3.  Here's the output from the CLI:

~/Documents/RaspPi --> sudo etcher ./2018-03-13-raspbian-stretch.zip
? Select drive /dev/disk1 (15.9 GB) - SD Card Reader
? This will erase the selected drive. Are you sure? Yes
You don't have access to this resource

Please ensure you have necessary permissions to access this resource

And then from the DTP app:

cations/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 {"stack":"Error: EACCES: permission denied, open '/dev/rdisk1'","message":"EACCES: permission denied, open '/dev/rdisk1'","code":"EACCES"}

(There's more, but that seemed most relevant.)  

Disk is SanDisk Ultra Plus 16GB.

Anything else I can add?

JulianNorton, 2018-04-16:
On windows 10: 
Upon beginning the flash I get the windows error:
```
You're not authorized to perform this operation
```
Then immediately the Etcher error:
```
Oops, seems something went wrong.
```

1. Error for both 1.2.1 and 1.3.1.
2. I also tried "running as an administrator" as 1.3.1 and same error.
3. Rebooted, same error in 1.3.1
3. Enabling `unsafe mode` resulted in the same error.

JulianNorton, 2018-04-17:
Want to recommend reopening this issue and tag it as Windows specific. I flashed on Mac with no problem.

JulianNorton, 2018-04-17:
@jhermsmeier I recommend reopening this issue, or creating a new issue, to clarify the error message. I believe the root cause of this is the SD card not staying unmounted.

raydot, 2018-04-18:
I flashed on Windows with no problem but couldn't get it working on Mac...could it be make/model of computer?

rahul260792, 2018-04-26:
I had the same issue on Windows 10 I fixed the issue by disabling Windows Defender as it was not allowing Etcher to flash the image in micro sd card.

fwcd, 2018-05-09:
@rahul260792 That solved it, thank you!

lfg6000, 2018-06-02:
I had this issue when trying to flash raspbian to a 32gb sandisk sd card on Win10
  You're not authorized to perform this operation
  Please ensure you have necessary permissions for this task

Fixed by temporarily turning off Win 10 Controlled File Access
- open windows defender
- open virus & threat protection
- open ransomware protection
- turn off controlled file access
- Etcher was able to flash the raspbian image to the sd card
- turn controlled file access back on

marcoamm, 2018-07-19:
@lfg6000 Worked for me, and the solution makes sense. Thanks!

CapitanLab, 2018-08-04:
I turn off all antivirus etc. and it works well.

DubTech247, 2018-09-11:
> I encountered the same problem in Win 7, and it turned out to be caused by Acronis True Image's 'active protection' function. Turn off the function and I can run the flash now.

@nestser 
i love you and wish to have your babies

drgruney, 2018-09-23:
I gave Windows defender an application exception for Etcher in Controlled Folder settings :D

consolacion, 2018-10-19:
none of this helped, so I was forced to go back to Win32 Disk Imager

klccjeremy, 2018-10-24:
works well before i updated to V1.4.5, after update had the same error msg

jjahn504, 2019-02-01:
I encountered the same problem in Win 10, Etcher1.4.9, and I turned off the **'Windows Defender**->real time service' and I can run the flash now.

UmerFarooqBiomedical, 2019-02-20:
> I had this issue when trying to flash raspbian to a 32gb sandisk sd card on Win10
> You're not authorized to perform this operation
> Please ensure you have necessary permissions for this task
>
> Fixed by [...]

Thank you so much, Man. I spent my whole day under stress and tried 2 different cards and faced the same problem, again and again, a hundred time till I find your comment. This solved my problem. A lot-lot-lot of thank. Love from Pakistan.

DubTech247, 2019-02-20:
Turns out it was some sort of drive protection feature from a recently
installed Acronis program, switched it off and it worked fine

lurch, 2019-02-21:
@thundron Sounds like it might be useful to include these various workarounds into the docs or FAQ?

haleypearse, 2020-03-14:
I solved this on a mac by opening with sudo from the terminal by typing 
sudo /Applications/Etcher.app/Contents/MacOS/Etcher

bestK, 2020-03-22:
update to last version to fixed in MacOS catalina

saharshtapi, 2020-03-25:
I had the same issue. Go to windows security > virus and threat > under ransomware click on manage ransomware protection and turn off control folder access. that's it