balena-io/etcher #1962: Figure out why maxBuffer was exceeded in the first place

Shou, 2018-01-08:
@jviotti posted in https://github.com/resin-io/etcher/pull/1961

> Can we research why this only happens on Linux? Is it because that platform has a much lower max buffer limit? What are the default limits in the other OSes? Are we sure that we can't ever reach that limit even with a very big image or very slow drive? What things are printed to stdout by the child writer?

jhermsmeier, 2018-01-08:
Some testing reveals that `node-ipc` is indeed the only thing writing to stdout / stderr in this case; see https://github.com/resin-io/etcher/pull/1961#issuecomment-356032203

closed 2018-01-08 by jhermsmeier

lurch, 2018-01-09:
https://github.com/jorangreef/sudo-prompt/issues/66 also has some details about why this only affects Linux.