balena-io/etcher #1964: TrueOS fails to burn to USB

kenny-wettlaufer, 2018-01-08:
- **Etcher version:** 1.3.0-1 is what it says I have installed BUT the AUR version is 1.2.1-1 
- **Operating system and architecture:** Artix Linux arch variant x64 OpenRC init
- **Image flashed:** TrueOS stable & unstable
- **Do you see any meaningful error information in the DevTools?** ummm no..

I'm downgrading to the 1.2.1-1 release to see if that helps... TrueOS keeps failing to burn to USB, it just gives me a basic error "oops, something went wrong" I wish I knew how to get more info about this error but I'm not really sure... How can I determine whether its just the .img or not?

kenny-wettlaufer, 2018-01-08:
[image]

jhermsmeier, 2018-01-08:
Hi @kenny-wettlaufer, downloading the image at the moment; will have a look shortly.
We axed the 1.3.0 release, because we missed a severe issue on Linux, so if you can downgrade to 1.2.1 for now that will probably get you there.

kenny-wettlaufer, 2018-01-08:
Thank you jhermsmeier, I couldn't understand why I had 1.3 and the AUR had 1.2.1. I am going to attempt it again now.

kenny-wettlaufer, 2018-01-08:
It burns OK now, sorry to have troubled you guys, love Etcher! Keep up the good work!

jhermsmeier, 2018-01-08:
I've just run this image through Etcher v1.3.0 on Mac OS; and it works fine, so this is issue #1955 – which is that bug on Linux. Marking it as duplicate now; the next release will fix this.

jhermsmeier, 2018-01-08:
Duplicate of #1955

closed 2018-01-08 by jhermsmeier

kenny-wettlaufer, 2018-01-08:
Thanks for your help!!! Cheers!