balena-io/etcher #1969: Etcher doesnt recognize USB drive

alejandritoqz, 2018-01-10:
- **Etcher version:**
1.2.1

- **Operating system and architecture:**
mac OS High Sierra version 10.13.2

- **Image flashed:**
ubuntu-16.04.3-desktop-amd64.iso

- **Do you see any meaningful error information in the DevTools?**
No. And my laptop is detecting the external drive. Command run: 
diskutil list
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *4.0 TB     disk2
   1:                        EFI EFI                     314.6 MB   disk2s1
   2:       Microsoft Basic Data AQZ007                  4.0 TB     disk2s2

lurch, 2018-01-10:
Have you tried enabling Etcher's "unsafe mode" in the settings? Etcher might be mis-identifying such a large drive as a "System Drive" (which we hide by default).

peterjuras, 2018-01-31:
Same problem here with 1.3.1

unsafe mode does not help, it lists no drives as connected, but I see the drives in Finder, terminal, etc.

mac OS High Sierra version 10.13.3

peterjuras, 2018-01-31:
the device reappeared after a reboot, still not optimal though :/

lurch, 2018-01-31:
@peterjuras What do you mean by "still not optimal though" ?

Could you install https://github.com/resin-io-modules/drivelist and let us know what that reports?

peterjuras, 2018-01-31:
Sorry for not being more specific, it was the second or third time that it didn't list the drive, therefore while a reboot fixed the issue, it still worries me that it just won't work after a reboot in some point in the future.

I'll post the reports soon

peterjuras, 2018-01-31:
This is the current output, however the device is also listed in Etcher now. I'll repost when I hit the bug again.

```js
[ { enumerator: 'diskutil',
    busType: 'UNKNOWN',
    busVersion: '0.0',
    device: '/dev/disk0',
    raw: '/dev/rdisk0',
    description: 'APPLE SSD AP0512J',
    error: null,
    size: 500277792768,
    blockSize: null,
    logicalBlockSize: null,
    mountpoints: [],
    isReadOnly: false,
    isSystem: true,
    isVirtual: null,
    isRemovable: null,
    isCard: null,
    isSCSI: null,
    isUSB: null,
    isUAS: null },
  { enumerator: 'diskutil',
    busType: 'UNKNOWN',
    busVersion: '0.0',
    device: '/dev/disk1',
    raw: '/dev/rdisk1',
    description: 'APPLE SSD AP0512J',
    error: null,
    size: 499313172480,
    blockSize: null,
    logicalBlockSize: null,
    mountpoints: [ [Object], [Object] ],
    isReadOnly: false,
    isSystem: true,
    isVirtual: null,
    isRemovable: null,
    isCard: null,
    isSCSI: null,
    isUSB: null,
    isUAS: null },
  { enumerator: 'diskutil',
    busType: 'UNKNOWN',
    busVersion: '0.0',
    device: '/dev/disk2',
    raw: '/dev/rdisk2',
    description: 'STORAGE DEVICE',
    error: null,
    size: 15931539456,
    blockSize: null,
    logicalBlockSize: null,
    mountpoints: [ [Object] ],
    isReadOnly: false,
    isSystem: false,
    isVirtual: null,
    isRemovable: null,
    isCard: null,
    isSCSI: null,
    isUSB: null,
    isUAS: null } ]
```

jhermsmeier, 2018-01-31:
@peterjuras If it stops working at some random point in time, I think this might have more to do with the operating system than Etcher. Which device do you have? Macbook Air / Pro / ...?

peterjuras, 2018-01-31:
[MacBook Pro 13 inch with Touchbar](https://www.apple.com/shop/buy-mac/macbook-pro)

timrowledge, 2018-02-20:
Just had this problem (iMac/High Sierra), unsafe mode allowed detection of the connected usb/ssd. it seems to have written correctly after that.

jhermsmeier, 2018-03-26:
@timrowledge if unsafe mode made it show up, and it's an SSD then that's the expected behavior for that version – we've changed it recently to show anything that's not an internal storage device though, so that particular problem should go away in the next release

jhermsmeier, 2018-03-26:
Closing this for now, as this should be fixed by https://github.com/resin-io-modules/drivelist/pull/253 which made it into Etcher in https://github.com/resin-io/etcher/pull/2137 and will be released in v1.4.0 soon – feel free to reopen should this arise again

closed 2018-03-26 by jhermsmeier

timrowledge, 2018-03-26:
OK; I'll let you know if there are subsequent problems.

wgriffa757, 2018-12-10:
Hi I am using Etcher version 1.4.8 on a MacBook Pro with light bar. I have attached an external Samsung usb DVD writer. Etcher sees the drive but it thinks it is locked. Disk utility also see it and I wonder if that is why Etcher thinks the drive is locked. I tried the unsafe mode, it did not help. 
Please help.

lurch, 2018-12-11:
@wgriffa757 see #2086

jakramer-aits, 2020-09-08:
My Win10 Dell Precision has a built in SD slot. Etcher could not find the drive there. It occurred to me that Etcher is looking for drives on the USB ports. Plugged a SD adapter into the USB port and Etcher immediately recognized that the SD card was there. Completed the flash and all is good.

Etcher version 1.5.107, Win10 Pro for Workstations, 19041.450