balena-io/etcher #1972: Make electron develop throws errors etcher

aamm132, 2018-01-12:
- **Etcher version:1.21**
- **Operating system and architecture:Windows 7**
I am attempting to open the testing suite, however whenever I do I get an error I cannot trace. I have followed the contributing page to the best of my ability but every time I run "make electron develop" I get a log that looks like the file attached. I am not sure where the error is or how to fix it. Please let me know what I can do to fix this, and thanks for any help provided.
[2018-01-12T22_11_35_193Z-debug.log](https://github.com/resin-io/etcher/files/1627728/2018-01-12T22_11_35_193Z-debug.log)

lurch, 2018-01-12:
I see you've got `3 info using node@v9.4.0` in your logfile. Could you try gain with NodeJS v6? (which is the version of Node we use on our CI servers - https://github.com/resin-io/etcher/blob/master/appveyor.yml#L24 )

The line that's failing is
```
129 silly lifecycle winusb-driver-generator@1.1.1~install: Args: [ '/d /s /c', 'prebuild-install || npm run rebuild' ]
```
(i.e. try to install the prebuilt version of `winusb-driver-generator`, and if that doesn't work rebuild `winusb-driver-generator` from source)
I believe the `prebuild-install` part is failing because `winusb-driver-generator` doesn't have any prebuilds available for NodeJS v9 https://github.com/resin-io-modules/winusb-driver-generator/releases and perhaps the `rebuild` part is failing because you don't have the necessary compilers or headers installed? 
@jviotti will be able to tell us more... :wink:  (he wrote the winusb-driver-generator module)

Oh! I've just spotted that https://github.com/resin-io-modules/winusb-driver-generator/blob/master/README.md#installation says "you can only build this module using Visual Studio 2013, and can't target a Node v9.0.0 ABI" - if I remember correctly there are also other parts of Etcher that need to be compiled with Visual Studio 2015, which I think is why we _need_ to use prebuilds for winusb-driver-generator.

jviotti, 2018-01-13:
Yeah, building libwdi (a dependency of `winusb-driver-generator`) is
quite a pain, since it needs Visual Studio 2013, Node.js <9.0.0, the
Windows Driver Development Kit, and some tweaks on the host OS to be
able to compile correctly (see the appveyor.yml file in
`winusb-driver-generator` to see how we prepare the environment).

lurch, 2018-01-13:
@jviotti Out of curiosity, how come https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig talks about using Visual Studio 2017? :confused:

aamm132, 2018-01-15:
Hmm, so I tried running the command npm install --save winusb-driver-generator (after switching to the node version 6.10.3) and I got this log out: 
[2018-01-15T18_04_05_724Z-debug.log](https://github.com/resin-io/etcher/files/1632606/2018-01-15T18_04_05_724Z-debug.log) 
@lurch mentioned there could be headers and compilers needed to run this, is Visual Studio 2013 one of the solutions to my problem? If so, is there a free download of VS2013 somewhere? I tried running these commands in git-cmd.exe, git bash, and even Visual Studio 2017 (which did not even recognize the npm command). 
Please let me know what step(s) I am missing, and thanks once again for any help! :grin:

lurch, 2018-01-15:
@aamm132 Your logfile still says:
```
2 info using npm@5.6.0
3 info using node@v8.9.4
```
?

aamm132, 2018-01-16:
Sorry about that, I thought I had switched but it turns out it still defaulted to that version. I was able to get past this issue, but ran into another one -- i'll post it in another issue. Thanks again.

closed 2018-01-16 by aamm132