balena-io/etcher #1974: Npm start & test

aamm132, 2018-01-16:
Etcher version:1.21
Operating system and architecture:Windows 7

I am trying to run these commands in Visual Studio (2017, it may be that I need to be using a different version), and I get two different results when running. For npm test, I get this output in the package manager console 

npm.cmd : Makefile:157: No code-sign certificate found (CSC_LINK is not set)
At line:1 char:4
+ npm <<<<  test
    + CategoryInfo          : NotSpecified: (Makefile:157: N...INK is not set):String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
Makefile:159: No code-sign certificate password found (CSC_KEY_PASSWORD is not set)

Makefile:175: No Sentry token found (ANALYTICS_SENTRY_TOKEN is not set)
Makefile:181: No Mixpanel token found (ANALYTICS_MIXPANEL_TOKEN is not set)

eslint lib tests scripts bin versionist.conf.js
sass-lint lib/gui/scss
cpplint --recursive src
/bin/bash: cpplint: command not found

make: 
*** [lint-cpp] Error 127

npm
 
ERR!
 Test failed.  See above for more details.

For npm start, it runs seemingly fine but get an empty etcher window that looks like this: 

[image]

Based off the earlier test output, I would like to think that I may be missing something in my makefile, but I am not sure. Any help provided would be appreciated!

jhermsmeier, 2018-01-16:
You'll need to have Python 2.7 installed, and then run `pip install -r requirements.txt` in the Etcher repository; detailed info on running / building it locally can be found in [docs/CONTRIBUTING.md](https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md)

aamm132, 2018-01-16:
I tried running it in cmd.exe, and got this line of text: 
Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
I did install Python 2.7, and set pip to my environments path variable (if that helps in figuring out my issue).

jhermsmeier, 2018-01-16:
Are you running that in the Etcher repository root? Because the [requirements.txt](https://github.com/resin-io/etcher/blob/master/requirements.txt) is definitely there :)

aamm132, 2018-01-17:
Whoops, running it in the etcher repository seems to have worked, but then I run into another problem (that produces the same results as mentioned in my first comment), saying the package.json does not exist. This is the debug log that is produced when running npm test: 
[npm-debug.log](https://github.com/resin-io/etcher/files/1639446/npm-debug.log)
I checked the files and the package.json file was in the etcher folder (as was the requirements.txt!). Did I perhaps miss another step on the [docs/CONTRIBUTING.md](https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md) page?

lurch, 2018-01-17:
`6 verbose cwd C:\Users\MesserA` seems to be your problem? It's therefore complaining that `C:\Users\MesserA\package.json` doesn't exist.
Try `cd`-ing to e.g. your `C:\Users\MesserA\etcher` directory before running `npm test`.

> ran "npm test" (without using make electron-develop)

I think that won't work - you need to run `npm test` _after_ running `make electron-develop`.

aamm132, 2018-01-17:
I was in the etcher directory when I ran npm test, but this time I ran it in cmd.exe and got this output:

The specified module could not be found.
\\?\C:\Users\MesserA\Desktop\etcher\node_modules\lzma-native\binding\lzma_native
.node
Error: The specified module could not be found.
\\?\C:\Users\MesserA\Desktop\etcher\node_modules\lzma-native\binding\lzma_native
.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)

    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
...
...
It kinda repeats like this for many lines until it says:
make: *** [test-gui] Error 1
npm Err! Test Failed. See above for more details.

So something a little different than when I ran this command in the VS package manager console (when I do it there I get the same output as my original comment when I am still in the etcher directory), though I am still unsure how to fix it. Thanks again for the help.

aamm132, 2018-01-18:
So I ran npm install --save lzma-native (from their own git page) since that's where I thought I was running into issues and got a somewhat different output:

Error: The system cannot find message text for message number 0x%1 in the messag
e file for %2.
\\?\C:\Users\MesserA\Desktop\etcher\node_modules\lzma-native\binding\lzma_native
.node

So, something a little different, but not sure if it is any progress getting these commands to work.

lurch, 2018-01-19:
There shouldn't be any need to manually `npm install` any modules - that's what the `make electron-develop` step is for. I'll attempt to recreate the problems you're having in a fresh Windows7 VM sometime next week.

aamm132, 2018-01-19:
Okay well here's the process I went through for your convenience,
1. git cloned etcher.
2. Following the contributing page, I installed requirements.txt, and windows-build-tools and any prerequisites 
3. Ran make electron-develop in cmd.exe
4. npm installed lzma-native (tried uninstalling it too, but didn't seem to do anything)
5. Ran npm test in cmd.exe and got the following output: 

> etcher@1.3.0 test C:\Users\MesserA\Desktop\etcher
> make lint test sanity-checks

Makefile:157: No code-sign certificate found (CSC_LINK is not set)
Makefile:159: No code-sign certificate password found (CSC_KEY_PASSWORD is
et)
Makefile:175: No Sentry token found (ANALYTICS_SENTRY_TOKEN is not set)
Makefile:181: No Mixpanel token found (ANALYTICS_MIXPANEL_TOKEN is not set
eslint lib tests scripts bin versionist.conf.js
sass-lint lib/gui/scss
cpplint --recursive src
Done processing src\os\elevate.h
Done processing src\utils\v8utils.cpp
Done processing src\os\win32\elevate.cpp
Done processing src\elevator_init.cpp
Done processing src\utils\v8utils.h
node scripts/html-lint.js
Scanning...
Passed
codespell \
                --dictionary - \
                --dictionary dictionary.txt \
                --skip *.gz,*.bz2,*.xz,*.zip,*.img,*.dmg,*.iso,*.rpi-sdcar
Store,*.dtb,*.dtbo,*.dat,*.elf,*.bin,*.foo,xz-without-extension \
                lib tests docs scripts Makefile *.md LICENSE
electron-mocha --recursive --reporter spec --renderer tests/gui

%csdk %cscanner:init%c +0ms color: darkorchid color: inherit color: darkor
ap {
  domain: null,
  _events: Map {},
  _eventsCount: 0,
  _maxListeners: undefined,
  options:
   Map {
     standard: Map { includeSystemDrives: false },
     usbboot: Map { readFile: [Function: callIntoRenderer] } },
  isScanning: false,
  adapters: Map {},
  onDevices: [Function: callIntoRenderer],
  onError: [Function: callIntoRenderer],
  setMaxListeners: [Function: callIntoRenderer],
  getMaxListeners: [Function: callIntoRenderer],
  emit: [Function: callIntoRenderer],
  addListener: [Function: callIntoRenderer],
  on: [Function: callIntoRenderer],
  prependListener: [Function: callIntoRenderer],
  once: [Function: callIntoRenderer],
  prependOnceListener: [Function: callIntoRenderer],
  removeListener: [Function: callIntoRenderer],
  removeAllListeners: [Function: callIntoRenderer],
  listeners: [Function: callIntoRenderer],
  listenerCount: [Function: callIntoRenderer],
  eventNames: [Function: callIntoRenderer] }
%csdk %csubscribe%c +15ms color: darkorchid color: inherit color: darkorch
omain: null,
  _events: { devices: [Function: callIntoRenderer] },
  _eventsCount: 1,
  _maxListeners: undefined,
  id: 'standard',
  devices: [],
  setMaxListeners: [Function: callIntoRenderer],
  getMaxListeners: [Function: callIntoRenderer],
  emit: [Function: callIntoRenderer],
  addListener: [Function: callIntoRenderer],
  on: [Function: callIntoRenderer],
  prependListener: [Function: callIntoRenderer],
  once: [Function: callIntoRenderer],
  prependOnceListener: [Function: callIntoRenderer],
  removeListener: [Function: callIntoRenderer],
  removeAllListeners: [Function: callIntoRenderer],
  listeners: [Function: callIntoRenderer],
  listenerCount: [Function: callIntoRenderer],
  eventNames: [Function: callIntoRenderer] }
%csdk %csubscribe%c +6ms color: darkorchid color: inherit color: darkorchi
main: null,
  _events: { devices: [Function: callIntoRenderer] },
  _eventsCount: 1,
  _maxListeners: undefined,
  id: 'usbboot',
  blobCache: {},
  progress: {},
  devices: [],
  setMaxListeners: [Function: callIntoRenderer],
  getMaxListeners: [Function: callIntoRenderer],
  emit: [Function: callIntoRenderer],
  addListener: [Function: callIntoRenderer],
  on: [Function: callIntoRenderer],
  prependListener: [Function: callIntoRenderer],
  once: [Function: callIntoRenderer],
  prependOnceListener: [Function: callIntoRenderer],
  removeListener: [Function: callIntoRenderer],
  removeAllListeners: [Function: callIntoRenderer],
  listeners: [Function: callIntoRenderer],
  listenerCount: [Function: callIntoRenderer],
  eventNames: [Function: callIntoRenderer] }

  Browser: DriveSelector
    DriveSelectorController
      .memoizeImmutableListReference()
      events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: write EIO
    at exports._errnoException (util.js:1018:11)
    at WriteStream.Socket._writeGeneric (net.js:711:26)
    at WriteStream.Socket._write (net.js:730:8)
    at doWrite (_stream_writable.js:331:12)
    at writeOrBuffer (_stream_writable.js:317:5)
    at WriteStream.Writable.write (_stream_writable.js:243:11)
    at WriteStream.Socket.write (net.js:657:40)
    at Socket.ondata (_stream_readable.js:555:20)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
make: *** [test-gui] Error 1
npm ERR! Test failed.  See above for more details.

Sorry it is kinda long, but that's just to see if you get something similar.
Thanks again for the help & responses.

aamm132, 2018-02-07:
I decided to re-do the steps by running a git clean and git reset, and things seemed to work. Thanks!

closed 2018-02-07 by aamm132