balena-io/etcher #1982: `npm test` Fails

Coder-256, 2018-01-21:
- **Etcher version:** e1ddbd3
- **Operating system and architecture:** macOS High Sierra 10.13.2, tested on Node v6.12.3 and v9.4.0, NPM 5.6.0, Python 2.7.14, pip 9.0.1
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

I just ran:

```
git clone 'https://github.com/resin-io/etcher.git'
cd etcher
pip2 install -r requirements.txt
make electron-develop
npm test
```

Everything = works great until this happens:

```
$ npm test

> etcher@1.3.0 test /Users/<redacted>/Downloads/etcher-sandbox/etcher
> make lint test sanity-checks

Makefile:151: No code-sign identity found (CSC_NAME is not set)
Makefile:175: No Sentry token found (ANALYTICS_SENTRY_TOKEN is not set)
Makefile:181: No Mixpanel token found (ANALYTICS_MIXPANEL_TOKEN is not set)
eslint lib tests scripts bin versionist.conf.js
sass-lint lib/gui/scss
cpplint --recursive src
Done processing src/utils/v8utils.cpp
Done processing src/os/elevate.h
Done processing src/elevator_init.cpp
Done processing src/os/win32/elevate.cpp
Done processing src/utils/v8utils.h
...
./scripts/ci/ensure-all-node-requirements-available.sh
./scripts/ci/ensure-staged-sass.sh
Makefile:151: No code-sign identity found (CSC_NAME is not set)
Makefile:175: No Sentry token found (ANALYTICS_SENTRY_TOKEN is not set)
Makefile:181: No Mixpanel token found (ANALYTICS_MIXPANEL_TOKEN is not set)
node-sass lib/gui/scss/main.scss > lib/gui/css/main.css
module.js:689
  return process.dlopen(module, path.toNamespacedPath(filename));
                 ^

Error: The module '/Users/<redacted>/Downloads/etcher-sandbox/etcher/node_modules/node-sass/vendor/darwin-x64-59/binding.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 59. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (module.js:689:18)
    at Module.load (module.js:573:32)
    at tryModuleLoad (module.js:513:12)
    at Function.Module._load (module.js:505:3)
    at Module.require (module.js:604:17)
    at require (internal/module.js:11:18)
    at module.exports (/Users/<redacted>/Downloads/etcher-sandbox/etcher/node_modules/node-sass/lib/binding.js:19:10)
    at Object.<anonymous> (/Users/<redacted>/Downloads/etcher-sandbox/etcher/node_modules/node-sass/lib/index.js:14:35)
    at Module._compile (module.js:660:30)
    at Object.Module._extensions..js (module.js:671:10)
make[1]: *** [sass] Error 1
make: *** [sanity-checks] Error 2
npm ERR! Test failed.  See above for more details.
```

I have no idea why this is not working because I have all the dependencies installed and I am running from a clean clone. As mentioned, I have tested this on Node v6.12.3 and v9.4.0.

lurch, 2018-01-21:
AFAICT, it looks like there's a mis-match between the version of NodeJS that `node_modules/node-sass` got installed with (`NODE_MODULE_VERSION 54`), and the version of NodeJS that's being used when you're running `npm test` (`NODE_MODULE_VERSION 59`).

Perhaps re-running `make electron-develop` will fix things?

Coder-256, 2018-01-21:
@lurch Well that's what I had gathered from the error message. Rerunning `make electron-develop` does not help (it's what I ran right before `npm test`. Interestingly enough, `make test` succeeds (although it does not run all tests I think).

lurch, 2018-01-22:
> Interestingly enough, `make test` succeeds (although it does not run all tests I think).

Actually, `make test` _does_ run all the tests, but `npm test` also runs additional checks. Here's what's happening... (in case it helps you track down what's going wrong)
1. `npm test` invokes `make lint test sanity-checks`
1. From your output above, the `lint` and `test` targets are running fine, it's the `make sanity-checks` which is failing
1. One of the sub-tasks of the `sanity-checks` target is running `./scripts/ci/ensure-staged-sass.sh` and it's this that is failing
1. `./scripts/ci/ensure-staged-sass.sh` in turn actually runs `make sass`
1. The `sass` target then runs `node-sass lib/gui/scss/main.scss > lib/gui/css/main.css`, and this is the step that's _actually_ resulting in the error.

So to debug your problem, could you try running just `make sass` in your terminal to see if that gives you the same error?
And if you sprinkle `node --version` in a few different places, that might help you see which node version is getting called by which node script? Out of curiosity, how exactly are you testing with both Node v6.12.3 and v9.4.0?

As the tests themselves all work (`make test`), I'm going to downgrade the priority of this issue.

Coder-256, 2018-01-22:
You're right, the problem was with sass. But now I'm getting another issue, this time with `./scripts/ci/ensure-staged-shrinkwrap.sh` (which is called by `npm test`). I tried running the following commands:

```
$ ./scripts/ci/ensure-staged-shrinkwrap.sh
node version: v9.4.0
npm version: 5.6.0
npm notice created a lockfile as npm-shrinkwrap.json. You should commit this file.

> etcher@1.3.0 postshrinkwrap /Users/jacob/Git/etcher
> node ./scripts/clean-shrinkwrap.js

Cleaning shrinkwrap...
 - Removing "7zip-bin-mac" from "etcher"

[OK] Wrote shrinkwrap file to ../npm-shrinkwrap.json
There are unstaged npm-shrinkwrap.json changes. Please commit the result of:

    npm shrinkwrap --dev

diff --git a/npm-shrinkwrap.json b/npm-shrinkwrap.json
index 151a8a5..28919b4 100644
--- a/npm-shrinkwrap.json
+++ b/npm-shrinkwrap.json
@@ -1,6750 +1,10281 @@
 {
   "name": "etcher",
   "version": "1.3.0",
+  "lockfileVersion": 1,
+  "requires": true,
   "dependencies": {
+    "7zip-bin": {
...
$ git add .
$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

	modified:   npm-shrinkwrap.json

$ ./scripts/ci/ensure-staged-shrinkwrap.sh
node version: v9.4.0
npm version: 5.6.0
npm notice created a lockfile as npm-shrinkwrap.json. You should commit this file.

> etcher@1.3.0 postshrinkwrap /Users/jacob/Git/etcher
> node ./scripts/clean-shrinkwrap.js

Cleaning shrinkwrap...
 - Removing "7zip-bin-mac" from "etcher"

[OK] Wrote shrinkwrap file to ../npm-shrinkwrap.json
There are unstaged npm-shrinkwrap.json changes. Please commit the result of:

    npm shrinkwrap --dev

```

I don't know why there was a new shrinkwrap created in the first place because I am on a fresh clone. But for whatever reason, I think `make electron-develop` creates a new one (I think it has something to do with the `"lockfileVersion"`), and it `npm test` is refusing to work until I commit the new version.

Here is the [diff of the new shrinkwrap file](https://gist.github.com/Coder-256/e380920093e899bbf64f3ef212c0d15e).

lurch, 2018-01-22:
Looks like the difference in `npm-shrinkwrap.json` is created because you're running NPM v5 whereas the version of the file in the repo was created with NPM v3. I _suspect_ if you use NPM v3 then everything will be fine.

Given that `make test` works (which runs all of Etcher's unit-tests), why are you trying to get `npm test` to work as well? Just curious :)

Coder-256, 2018-01-22:
Alright, that makes sense. I was running `npm test` because [it's what `CONTRIBUTING.md` says](https://github.com/resin-io/etcher/blob/ffbb62a280108e011c69fbbfa00573b239ff7efa/docs/CONTRIBUTING.md#testing).

closed 2018-01-22 by Coder-256

lurch, 2018-01-22:
Yeah, if you want to contribute to Etcher you really ought to be using NodeJS 6 and NPM 3 ;-)

lurch, 2018-01-22:
@jhermsmeier I wonder if we ought to add some kind of note about the restriction of only using NPM v3 to CONTRIBUTING.md ?