balena-io/etcher #1991: Error couldn't clean the drive...

cwsyt7, 2018-01-24:
- **Etcher version:**1.2.1
- **Operating system and architecture:** Windows 10 Home v. 1703, built-in SD card reader (Samsung 32 GB micro SD card in adapter)
- **Image  #flashed:** Retropie 4.3
- **Do you see any meaningful error information in the DevTools?**  Yes, although I'm not sure what it means.

I am trying to install the retropie onto my SD card, however Etcher keeps hanging on "starting" for a few seconds then erroring out (see below). So far I have tried to reinstall Etcher a couple times, tried to run Etcher as both administrator (I get the error) and not as the administrator (just hangs on "starting"). I tried cleaning the SD card with diskpart, however I keep getting this error. I can manually copy files onto the SD card from within windows, so the SD card isn't locked out.

Any advice that can be offered would be greatly appreciated.

Thank you.

Windows Error Message:

[image]

Etcher DevTools

[image]

jhermsmeier, 2018-01-24:
Hey, this is a known issue with Windows; the drive cleaning will be retried 5 times, but sometimes it takes more than that, or Windows won't do it at all (and would need a restart to do so). It's surprisingly fragile.

One way you can work around this is with Disk Management; first remove all volumes on the device there, and then retry flashing it.

jhermsmeier, 2018-01-24:
You also might want to try Etcher v1.3.1, which we've released yesterday

cwsyt7, 2018-01-25:
Thanks for the info, my google fu didn't turn up anything. So if I keep retrying it, eventually it will work?

cwsyt7, 2018-01-25:
Ok, I downloaded the updated version of Etcher, but now I'm getting a new error (same PC hardware, Retropie version).

Popup error window:

[image]

Code:

[image]

lurch, 2018-01-25:
Thanks @cwsyt7 . Please could you switch to the "Console" tab of the DevTools window, and copy'n'paste the whole thing here please?

cwsyt7, 2018-01-25:
Here is the console tab.

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.3.1, Type = nsis
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:02 GMT-0500 (Eastern Standard Time) Application start ({"packageType":"nsis","version":"1.3.1"})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:02 GMT-0500 (Eastern Standard Time) Checking for updates ({"currentVersion":"1.3.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 sdk start +869ms true
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:03 GMT-0500 (Eastern Standard Time) Update notification skipped ({"reason":"Latest version"})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:06 GMT-0500 (Eastern Standard Time) Open image selector
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:10 GMT-0500 (Eastern Standard Time) Select image ({"path":"C:\\Users\\Chris & Marie\\Downloads\\retropie-4.3-rpi2_rpi3.img.gz","archiveExtension":"gz","extension":"img","size":{"original":641640647,"final":{"estimation":true,"value":2249195520}},"hasMBR":true,"hasGPT":false,"partitions":[{"type":14,"id":null,"name":null,"firstLBA":8192,"lastLBA":124928,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":124928,"lastLBA":4390912,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 sdk stop +10s true
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:12 GMT-0500 (Eastern Standard Time) Progress (undefined): 0% at 0 MB/s (eta undefineds)
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:12 GMT-0500 (Eastern Standard Time) Flash ({"image":"C:\\Users\\Chris & Marie\\Downloads\\retropie-4.3-rpi2_rpi3.img.gz","drive":{"size":32010928128,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"adapter":"standard","isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"mountpoints":[{"path":"D:\\"}],"description":"SDHC Card","isUAS":false},"uuid":"25b0f12a-5499-47d5-8043-b16f2bca6560","unmountOnSuccess":true,"validateWriteOnSuccess":true})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Elevating command: C:\Users\Chris & Marie\AppData\Local\Programs\etcher\Etcher.exe C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Flash results Object
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Terminating IPC server
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:12 GMT-0500 (Eastern Standard Time) Child died unexpectedly ({"image":"C:\\Users\\Chris & Marie\\Downloads\\retropie-4.3-rpi2_rpi3.img.gz","drive":{"size":32010928128,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"adapter":"standard","isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"mountpoints":[{"path":"D:\\"}],"description":"SDHC Card","isUAS":false},"uuid":"25b0f12a-5499-47d5-8043-b16f2bca6560","unmountOnSuccess":true,"validateWriteOnSuccess":true})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:12 GMT-0500 (Eastern Standard Time) Open modal ({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 sdk start +160ms true
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:21 GMT-0500 (Eastern Standard Time) Modal accepted ({"value":true})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:21 GMT-0500 (Eastern Standard Time) Restart after failure
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 sdk stop +15s true
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:27 GMT-0500 (Eastern Standard Time) Progress (undefined): 0% at 0 MB/s (eta undefineds)
[... cut ...] C Card","isUAS":false},"uuid":"301d448c-325f-4d41-a0e0-8a1ab3f80f55","unmountOnSuccess":true,"validateWriteOnSuccess":true})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 Thu Jan 25 2018 07:09:27 GMT-0500 (Eastern Standard Time) Open modal ({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
C:\Users\Chris & Marie\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\…:55 sdk start +120ms true

cwsyt7, 2018-01-27:
Just wanted to let y'all know I was able to get it installed with Win32 disk imager, although if you are able to figure out what is going on with this one I would love to know, etcher was quite a bit more user friendly.

Thanks for the support.

Inphinity, 2018-01-28:
I'm consistently seeing this error on multiple USB drives once I've used Etcher to create a USB installer for Centos 7 minimal.

I have this error in my event viewer for VDS basic provider:
`Cannot zero sectors on disk \\?\PhysicalDrive2. Error code: 5@0101000F`

This error appears in groups of 5 (roughly 2 seconds apart) when caused by Etcher, and just once when caused by me using diskpart.

I have found the solution (at least for me) is to run 2 successive clean operations _before_ using etcher.

`DISKPART> list disk`
`DISKPART> select disk 2`
`DISKPART> clean`
`DISKPART> list disk`
`DISKPART> select disk 2`
`DISKPART> clean`
`DISKPART> exit`

manuti, 2018-01-29:
Possibly related to #1955 ? Maybe this card are "bricked" by the 1.3 error?

jhermsmeier, 2018-01-29:
@manuti #1995 is not anything that could brick cards, and is completely unrelated to this

jhermsmeier, 2018-01-29:
Diskpart on Windows is a fickle beast, and has become less reliable in newer Windows versions for some reason; rarely works first time around – which is why Etcher retries that 5 times, and hence you're seeing those groupings @Inphinity

lurch, 2018-02-28:
@cwsyt7 Could you please test this version of Etcher to see if it fixes your problems? Thanks :)
https://resin-nightly-downloads.s3.amazonaws.com/etcher/2018-02-20/1.3.1%2B5be64e1/Etcher-Portable-1.3.1%2B5be64e1-x64.exe

toledanosjesus, 2018-03-08:
Hi,
I'm getting the same error. Using the portable version I got a different one:

[image]

Any help?

Cheers!

thor-schueler, 2018-03-21:
Note on the above, you will also get this error when you have bitlocker enabled (sigh). Not encrypting the media makes it read only, cleaning removes encryption, makes it read only. Got to find a device without bitlocker...

jhermsmeier, 2018-05-17:
Thanks @HennieLouw – will have a look at that shortly!

lurch, 2018-05-17:
Possibly similar to #1411 then?

ishanrakitha, 2018-05-20:
I was able to fix the issue by using the easeus partition master free edition. It will recreate the MBR for the drive.

HelpMeGitGud, 2018-06-11:
Pretty much same problem for me, etcher works initially but when trying to flash on top of another image diskpart doesn't seem to work. (neither from etcher nor from the command line) I thought it might be a problem with my usb stick as it is a little old but a new one (different model, bought today) had the same problem. Not sure how they do it but that easeus thing also solved the problem for me. Would be cool if their solution could be merged into etcher somehow but it doesn't seem to be open source. I could try to reproduce the issue another time if that helps.

jhermsmeier, 2018-06-11:
Thanks @ishanrakitha, @HelpMeGitGud – we've figured out the cause of it in this particular situation, and are fixing that soon as well; basically Windows doesn't consider a device "empty" if it doesn't have an empty MBR, so for a device to show up as "clean" on Windows, it actually needs a bit of formatting. If there's no empty partition table, or something Windows can't read / handle, it gets into this state.

ishanrakitha, 2018-06-11:
@HelpMeGitGud  try "SD Card Formatter". worked for me. @jhermsmeier glad you guys are looking into it.

Charios01, 2018-06-26:
Had the same problem, Etcher didnt write to the sd card and gave the error as above/couldnt clean and so on. Formatting the card was useless, also with Rufus. Solution was simple:
1. Formatting an a canon compact cam (A800)
2. Activating the service for virtual drives in the 'Dienste und Anwendungen' setting, that you get by the administrate function by clicking on 'Arbeitsplatz' with right mouse. My english is awful, so I use better the original windows titles in german...

MakiseKurisu, 2018-09-15:
Here is a workaround (for me at least) for "cannot zero sectors on disk" with error code 5@0101000F. All is done through diskpart so Etcher can incorporate this with your own diskpart script:
```
list disk #find out which disk you want to wipe
clean #throw out error, just a confirmation step but I always do this before apply workaround
convert gpt #disk in question is in MBR. I only burnt MBR image so I don't have a case where disk is in GPT
clean #this time you should be able to wipe the disk
```

closed 2018-10-15 by jviotti

Ramya1509, 2018-11-24:
[image]
why do i get this error when i try to flash???

lurch, 2018-12-03:
@Ramya1509 If you're getting that error with the latest version of Etcher, it might be worth opening a new issue.

flouck, 2021-04-22:
> 
> 
> Here is a workaround (for me at least) for "cannot zero sectors on disk" with error code 5@0101000F. All is done through diskpart so Etcher can incorporate this with your own diskpart script:
> 
> ```
> list disk #find out which disk you want to wipe
> clean #throw out error, just a confirmation step but I always do this before applying workaround
> convert gpt #disk in question is in MBR. I only burnt MBR image so I don't have a case where disk is in GPT
> clean #this time you should be able to wipe the disk
> ```
> 
> No reboot or unplug reqiured. By doing this Etcher will be able to clean the disk again and complete the burning process.

I tried but it did not work.
With latest Balena version it does not work either...
It is a SD card of course.
[image]

xmesaj2, 2023-04-12:
I just had the same problem and managed to finally fix it by formatting my stick with sdcard.org formatter v5
https://www.sdcard.org/downloads/formatter/

magnusquest, 2023-04-21:
@xmesaj2 this actually worked lol thanks!