balena-io/etcher #1992: Trying to install on a Raspberry Pi

wiggletech, 2018-01-24:
pi@raspberrypi:~ $ sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.UuS6yDNBF7/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data

Any idea?  Also, I had to instal dskmngr manually.  No big deal there though.
Thanks!

jhermsmeier, 2018-01-24:
Unless you've been trying to install Etcher, I think you might be asking in the wrong place?

jhermsmeier, 2018-01-24:
Oh, I see now – we currently don't have packages for ARMv7 (Raspberry Pis, etc) available at this time, so you'd have to build it yourself.

closed 2018-01-24 by jhermsmeier

wiggletech, 2018-01-24:
Yes.   I was trying to install the .deb package of Etcher on a pi.  The command I executed came from https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 <https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64>

Frank Rovito
Lighting Designer
Master Electrician / Technician
wiggletech@gmail.com <mailto:wiggletech@gmail.com>
216-212-7455
Frank Rovito Production and Design LLC.

Book Early, Book Often. ™

jhermsmeier, 2018-01-24:
Connects to #1688

wiggletech, 2018-01-24:
I thought that was what I was doing….  I am a tad more than a novice w/ linux and am educating myself.    Should I be installing a .deb package or should I be looking for source code from somewhere?  Please don’t give me all the answers, just a little direction.  Thanks!

Frank Rovito
Lighting Designer
Master Electrician / Technician
wiggletech@gmail.com <mailto:wiggletech@gmail.com>
216-212-7455
Frank Rovito Production and Design LLC.

Book Early, Book Often. ™

wiggletech, 2018-01-24:
Got it!   Thanks again!

Frank Rovito
Lighting Designer
Master Electrician / Technician
wiggletech@gmail.com <mailto:wiggletech@gmail.com>
216-212-7455
Frank Rovito Production and Design LLC.

Book Early, Book Often. ™

lurch, 2018-01-24:
If you're a Linux novice, then you might find trying to build Etcher for ARM a bit tricky. But if you're up for it, then see https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md

wiggletech, 2018-01-24:
Thanks!  Will do.

Frank Rovito
Lighting Designer
Master Electrician / Technician
wiggletech@gmail.com <mailto:wiggletech@gmail.com>
216-212-7455
Frank Rovito Production and Design LLC.

Book Early, Book Often. ™

jhermsmeier, 2018-01-24:
@lurch @wiggletech building Etcher on ARM is pretty broken at the moment, due to node-sass – so I wouldn't recommend trying it, unless you're willing to patch a few things in the code by hand – I've created #1993 to track these things, as we need to get them fixed soon to finally publish ARM builds, and for Etcher Pro.

wiggletech, 2018-01-24:
Ok…   I’ll wait then.  Thank you for helping.

Frank Rovito
Lighting Designer
Master Electrician / Technician
wiggletech@gmail.com <mailto:wiggletech@gmail.com>
216-212-7455
Frank Rovito Production and Design LLC.

Book Early, Book Often. ™

Technicavolous, 2018-05-10:
Any progress on this?

Technicavolous, 2018-05-10:
Any progress here?

jhermsmeier, 2018-05-11:
There's been some progress, there are still a thing or two to be done (see https://github.com/resin-io/etcher/issues/1993). These aren't actively worked on at the moment though.

Askmewho, 2018-12-09:
https://drive.google.com/file/d/1tbt4aQ2EGAKR6u4AhOVyXF9bfXJaDUhn/view?usp=sharing

I get something, but electron-builder didnt achieve a .deb because of architecture problem (fmp-linux its an x86 package and it doesnt have an arm one).

Askmewho, 2018-12-10:
You could at least run the program. Just unzip and run.

Botspot, 2020-01-09:
On the Raspberry Pi forums, a user named tinker2much managed to compile an [Etcher .deb for Raspberry Pi](https://www.raspberrypi.org/forums/viewtopic.php?f=62&t=255205&start=25#p1564958).
Installation is easy:
`wget https://github.com/futurejones/balena-etcher-arm/releases/download/v1.5.63/balena-etcher-electron_1.5.63_Raspbian_Buster_armhf.deb
sudo apt-get install ./balena-etcher-electron_1.5.63_Raspbian_Buster_armhf.deb`

lurch, 2020-01-09:
Just for reference, #2870 is the currently-open "Etcher on Raspberry Pi please" issue :wink: