balena-io/etcher #1999: apt-key not working

kenny-wettlaufer, 2018-01-25:
- **Etcher version:** none yet
- **Operating system and architecture:** Debian i686
- **Image flashed:** none
- **Do you see any meaningful error information in the DevTools?**

sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
[sudo] password for kzn: 
Executing: /tmp/apt-key-gpghome.4bNsB98ylZ/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data

kenny-wettlaufer, 2018-01-25:
ah nevermind, it seems to be working now. I tried it over & over again for a little while & it wasn't working, odd.

lurch, 2018-01-25:
Possibly a temporary problem with hkp://pgp.mit.edu:80 ? (which is obviously outside of our control anyway)

Glad to hear that it's working again :)

closed 2018-01-25 by lurch