balena-io/etcher #2007: ArchLinux32

kenny-wettlaufer, 2018-01-29:
- **Etcher version:** X 
- **Operating system and architecture:** Arch Linux 32-bit
- **Image flashed:** X
- **Do you see any meaningful error information in the DevTools?**

Hey everyone. There is now an "unofficial" 32-bit Arch Linux community at archlinux32.org, I noticed etcher is no longer supported for i686 in the AUR & am not sure if you guys maintain that AUR package or not & I figured you guys may have stopped supporting i686 on the AUR package because Arch Linux dropped i686, but its alive again - I don't know if you guys want to still maintain i686 or not but I figured I would just let you know. Thanks!

jhermsmeier, 2018-01-29:
Hey @kenny-wettlaufer, we're not maintaining the Etcher AUR packages, and haven't had contact with the author so far I believe, so it's not something we can change quickly at this time.

jhermsmeier, 2018-01-29:
Generally, we haven't discontinued 32bit support or anything, so that should be an easy one to fix :)

kenny-wettlaufer, 2018-01-29:
Oh awesome! I'm glad to hear you haven't dropped 32-bit, are you guys going to be making an AUR entry for Etcher? Or perhaps get Etcher onto the local archlinux32.org repos?