balena-io/etcher #2011: Etcher doesn't work on VoidLinux

lurch, 2018-01-31:
A user on gitter ( @tom-i ) has reported that whilst the AppImage runs on VoidLinux, it isn't able to flash an image.

We should investigate this.

jhermsmeier, 2018-01-31:
@tom-i which CPU arch and desktop are you running?

tom-i, 2018-01-31:
Ha, guys.. I used to use 1.1.2 version.
Today I downloaded 1.3.1 and it works like a charm..
So for VoidLinux users problem solved somehow by yours updates ;)
Please close this issue.

My configuration was:
uname -a
Linux PC 4.14.15_3 #1 SMP PREEMPT Sun Jan 28 22:13:16 UTC 2018 x86_64 GNU/Linux

lsb_release -a
LSB Version:	1.0
Distributor ID:	VoidLinux
Description:	Void Linux
Release:	rolling
Codename:	void

lurch, 2018-01-31:
That's great news, thanks for letting us know! :smiley:

closed 2018-01-31 by lurch

Jakecheslekov, 2018-02-09:
I'm running i686 Void, are you guys gonna be adding support for Void Linux i686 & x86_64?

lurch, 2018-02-10:
Just for clarity and to avoid any confusion:
If your kernel reports i686 then you're running in 32-bit mode and need to use the x86 version of Etcher
If your kernel reports x86_64 then you're running in 64-bit mode and need to use the x64 version of Etcher

@Jakecheslekov Does the x86 AppImage not work on i686 VoidLinux? I never got round to investigating this as @tom-i reported that the x64 AppImage worked fine on x86_64 VoidLinux.
Do you see any errors displayed if you run the x86 Etcher AppImage from a terminal window?