balena-io/etcher #2018: 1.3.1 just wrecked 2 SD cards

cloudmaestro, 2018-02-05:
Etcher version:1.3.1
Operating system and architecture:Win 10 Home 1709
Image flashed:3Gb image that flashed fine multiple times on 1.2.1
Do you see any meaningful error information in the DevTools? Not tried

Upgraded to 1.3.1. 1st SD card: Etcher stalled on "Starting". Waited 5 mins. Nothing so exited. Now card is wrecked. 2nd SD card - different make, same result. 2 16Gb U3 cards now unusable. Etcher has suddenly become too expensive for me to use. Shame. I loved it.

lurch, 2018-02-06:
I doubt the cards are really "wrecked" - they should be recoverable with https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

cloudmaestro, 2018-02-06:
Hi Andrew

Thanks for your email, I will try this. I really liked etcher and have
recommended it to others. I provide support to a number of companies
that use the Raspberry Pi for background music services. This requires
burning of 10s of SD cards up to a max of 100. I am particularly
interested in the multi-burner as the commercial units for this are
stupidly expensive and don't always do a good job as most of them seem
to use a Windows o/s. Quantities are not large enough to justify using
an external I know of several companies that would benefit from an
Etcher burner. There are a couple of tools I've found that are basically
a hack of WinDiskImager but they really don't cope with burning more
than 5-6 cards at a time.

I hadn't used Etcher for a while and was under the cosh to get a demo
unit out. So it threw me when the upgrade to 1.3.1 (which I just
accepted) caused an issue. I'm not super technical and did re-run a burn
with the chrome dev tools open. Again Etcher just stuck at the
"Starting" prompt and I couldn't see anything obvious in the dev tools
views to show where or why it was stuck. If I can help with debugging I
will as, up to now, I found Etcher to be fast and clean as a tool and,
as long as it's reliable, I will certainly be in line to buy a h/w
multi-burner when they are available and to recommend that to others.

Regards
Nigel

-- 

-------------------------
Nigel Allinson
m: +44 (0) 7913 963226
skype: nigel.allinson.ioh

 

Links:
------
[1]
https://github.com/resin-io/etcher/issues/2018#issuecomment-363360234
[2]
https://github.com/

lurch, 2018-02-06:
Thanks for your comments @cloudmaestro - I assume you've already signed up to the Etcher Pro mailing list? :)

I'm not sure why Etcher 1.3.1 isn't working for you - could you try running it as Administrator to see if that makes any difference?

jon4th4, 2018-02-10:
same here two flash sd 
[image]

jhermsmeier, 2018-06-05:
@cloudmaestro did you eventually have any luck restoring the cards? And have you had any issues like this with the newer Etcher versions (which also support flashing multiple devices simultaneously now)?

closed 2018-10-15 by jviotti