balena-io/etcher #2019: [Request] etcher flatpak via flathub

Tids, 2018-02-06:
Hi.
It would be nice to have a [flatpak](https://flatpak.org/) of etcher available on [flathub](https://github.com/flathub/flathub/wiki/App-Submission). So it would be included in many linux distros by default (and even more, when KDE discover also shipps flatpak support with the next release). That way we could just install it, without googling around how to get it.

lurch, 2018-02-06:
Do the current AppImages not work properly for you?
We're currently using [electron-builder](https://github.com/electron-userland/electron-builder) to create our Etcher packages (including AppImages, debs and rpms), and that doesn't [yet](https://github.com/electron-userland/electron-builder/issues/512) support creating flatpaks.

Last time I looked (which was admittedly quite a while ago) flatpaks were supported on fewer distros than AppImages. (e.g. flatpak is only for Ubuntu 16.04 and newer whereas our AppImages work on Ubuntu 12.04 and newer)
Also, it [looks like](https://flatpak.org/getting.html) flatpaks require you to install the `flatpak` command first, whereas AppImages "just run" with no other packages needing to be installed? \*shrug\*

Tids, 2018-02-06:
> Do the current AppImages not work properly for you?

I dont know if it works. Thing is that appimages are hard to manage. You need to download them from somewhere and do all that stuff manually. I really dont want to start with that. It seems that they also dont provide delta-updates or an updating service. And I dont want to download the whole "pack" just because there is a security update in some of its libs, or they simply dont ship security updates because of that reason.. So I would keep the AUR instead of appimages (and AUR is already a pain to use).

Yes flatpak requires the flatpak command and (if you want to) a graphical frontend. It also requires Internet

lurch, 2018-02-06:
Well I guess different people want different things - some people find the "download a single file and run it" approach much easier.

We'll consider adding flatpaks if / when electron-builder supports them.

Pointedstick, 2018-02-20:
AppImages work fine for me, but I too would like to see a Flatpak package on Flathub because of the same reason as @Tids: AppImage packages are a pain in the butt to manage on Linux. The usage model ("download from random website, put it wherever you want, double-click to open") is totally different from the typical Linux approach and doesn't integrate well with existing tools, workflows, or user expectations. Linux users are _heavily_ discouraged from downloading and running binaries from the Internet, in favor of using their package manager or graphical app store. Also the AppImage format is inherently un-user-friendly in that the user has to manually make the package executable first, which is an advanced task (even though it seems easy to people like us) that most regular users will fail at.

lurch, 2018-02-20:
> in favor of using their package manager

Just as an FYI, we *do* provide `.deb` and `.rpm` packages https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64
(but this obviously doesn't cover the plethora of different packaging formats that various Linux distros use)

> Also the AppImage format is inherently un-user-friendly in that the user has to manually make the package executable first, which is an advanced task (even though it seems easy to people like us) that most regular users will fail at.

That's why we distribute our Etcher AppImage files inside zipfiles - when the user extracts the AppImage from the zip, the executable bit is preserved (already set) :slightly_smiling_face:

HoboPrimate, 2018-03-24:
My comment on a duplicate issue:

> I believe flatpaking eletron apps is still a work in progress, but doing it would give the user advantages, such as peace of mind of it being sandboxed and allow them to update etcher easily. It would also make etcher be more discoverable if hosted on flathub.org.
> 
> Consider this just a small wish, as the appimage version seems to works fine!

kenny-w7, 2018-03-30:
I can't make AppImages work no matter what I do, a flatpak image would be nice, BUT....
Being able to build-from-source is always more preferable. Its a shame electron isn't supported by Void MUSL.

kenny-w1, 2018-04-12:
Hey guys, I found this on github, I wonder if this would help get etcher on flatpak? If electron is there then it should work.. right?
https://github.com/endlessm/electron-installer-flatpak

kenny-w1, 2018-04-12:
Flatpak on twitter posted this:
https://twitter.com/flatpakapps/status/887729426372415489

http://blog.manuq.com.ar/posts/building-electron-apps-offline-for-flathub/

dietrichstats, 2018-06-02:
> That's why we distribute our Etcher AppImage files inside zipfiles - when the user extracts the AppImage from the zip, the executable bit is preserved (already set) slightly_smiling_face

@lurch Exactly that is one of the reasons why it is not very intuitive, since every many developers that use AppImage "try" to make their use easier and with that they break with the expected mode. I've even seen installation script associated with an AppImage.

AppImage is an excellent portable format, in the sense of portable Windows applications, but lacking integration with the most popular software stores in Linux makes it a problem, as even users unfamiliar with Linux will understand the dynamics of a software stores. On the other hand, I consider it a mistake that integrated GPG signatures or appimaged are optional stuff.

lurch, 2018-06-03:
> integration with the most popular software stores in Linux

https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64
https://github.com/resin-io/etcher#redhat-rhel-and-fedora-based-package-repository-gnulinux-x86x64

Can't please all the people all the time...

dietrichstats, 2018-06-03:
> https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64
> https://github.com/resin-io/etcher#redhat-rhel-and-fedora-based-package-repository-gnulinux-x86x64
> 
> Can't please all the people all the time..

Dude, that was not necessary. You know well that I was talking about the AppImage package...

lurch, 2018-06-03:
Sorry, didn't mean to cause any offence. (I couldn't tell from your comment if you were already aware that Etcher is already packaged in deb and rpm repos)

jhermsmeier, 2018-06-05:
Thanks for all the info @Tids, @kenny-w1 – I think it's time to reevaluate our packaging choices and mechanisms. That'll likely take some time though, and the fragmentation among packaging schemes for Linux isn't helping.

lurch, 2018-06-05:
Obligatory XKCD: https://xkcd.com/927/  :grinning:

ghisvail, 2018-07-08:
I intend to give it a go (packaging `io.resin.etcher` as a Flatpak and distributing it via `flathub`).

jviotti, 2018-10-15:
Let's close this for now. Happy to take contributions

Jacalz, 2019-05-13:
Is there any news regarding this? I would absolutely love seeing Etcher on FaltHub and it would (in my opinion) be best if the core team took this on and updated it themselves. I don't mind if a user make makes the initial thing, but I think it would be best for the team to have it as a thing in their release process to update FlatHub too in order to make sure that it always is updated. 🙂

thundron, 2019-05-14:
No news yet, we currently have no plans to take etcher on flathub; if anyone feels like doing this, please do as that will be more than welcome, we also have people maintaining the `aur` package for Arch Linux and adding yet another package type feels a bit too crowded imho. We already have `AppImage`s and we would like that to stay the main cross-distro package when people need it, but again we're more than happy to take contributions if anyone feels like creating *and* maintaining a new package type!

ghisvail, 2019-05-14:
I am on it as we speak :-)

thundron, 2019-05-14:
@ghisvail If you need any help just ask!

A6GibKm, 2019-08-10:
I can also help with the creation of the flatpak manifesto.

ghisvail, 2019-08-12:
I have adapted the current Flatpak template for Electron apps but stumble upon the following issue when building etcher:

```
npm ERR! Error while executing:
npm ERR! /usr/bin/git ls-remote -h -t ssh://git@github.com/resin-io/node-usb.git
npm ERR! 
npm ERR! ssh: Could not resolve hostname github.com: Temporary failure in name resolution
npm ERR! fatal: Could not read from remote repository.
npm ERR! 
npm ERR! Please make sure you have the correct access rights
npm ERR! and the repository exists.
npm ERR! 
npm ERR! exited with error code: 128

npm ERR! A complete log of this run can be found in:
npm ERR!     /run/build/etcher/flatpak-node/npm-cache/_logs/2019-08-12T11_24_55_334Z-debug.log
Error: module etcher: Le processus fils s’est terminé avec le code 1
```

Looks like part of the build system still references the old path to the node-usb fork at `github.com/resin-io/node-usb.git`, instead of `github.com/balinea-io/node-usb.git`.

Besides, the build system should probably be using an https url for `node-usb` instead of the git+ssh protocol.

lurch, 2019-08-12:
@ghisvail Looks like https://github.com/balena-io-modules/node-raspberrypi-usbboot/blob/master/package.json#L31 might be to blame?
ping @zvin

A6GibKm, 2019-08-12:
@ghisvail, could you share you progress? Try applying a .patch for that url, e.g. https://github.com/flathub/org.gnu.emacs.

ghisvail, 2019-08-12:
> could you share you progress?

I can't get `etcher` to build offline with the error reported above.

> Try applying a .patch for that url

`node-raspberrypi-usbboot` gets pulled from npm so it's not as straightforward to patch as the main code base. I'd prefer the issue be fixed upstream instead of adding complexity to the Flatpak build system.

zvin, 2019-08-20:
@ghisvail this should be fixed in `v1.5.55`

mario-g98, 2019-12-18:
> No news yet, we currently have no plans to take etcher on flathub; if anyone feels like doing this, please do as that will be more than welcome, we also have people maintaining the `aur` package for [...]

I don't see why you aren't releasing a flathub version. AppImage is almost useless as nobody likes going to a website and downloading a file in order to install/update software. Flatpak is being used more and more and it's supported by all main distros, so by releasing a flatpak version you have everybody covered.

ghisvail, 2019-12-18:
> AppImage is almost useless as nobody likes going to a website and downloading a file in order to install/update software.

Go tell that to anyone using a Mac. Different formats for different use cases. 

> so by releasing a flatpak version you have everybody covered

It is also the case for AppImage. Enabling AppImage and snap artefacts from an Electron project is one declaration away in a file. Flatpak is yet to have this level of integration and is therefore more work to maintain comparatively.

Next time you approach upstream developers with a similar request, please use a different angle and a different set of arguments.

x80486, 2020-02-09:
Hey folks, I started this Flatpak adventure [here](https://github.com/x80486/flathub/tree/io.balena.etcher), but there is no way Yarn can pull `unbzip2-stream` from `etcher-sdk` (https://github.com/balena-io-modules/etcher-sdk/blob/master/package.json#L71).

Is it possible for you to change `unbzip2-stream` into a usual version range within `etcher-sdk`?

x80486, 2020-02-09:
In the meantime, building the Flatpak from `.deb` files work – while not optimal, it will let you use the Flatpak application in rolling releases, since the AppImage rarely runs without installing another gazillion on dangling dependencies :rofl:

If someone wants to take over this one, I can [finally send](https://github.com/flathub/flathub/pull/1363/) it to Flathub, try to get it done over there...and I can help from time to time, but I cannot be the (main) maintainer for this one :wink:

thundron, 2020-02-11:
It's the package we use to elevate the flashing process, you need that as well

x80486, 2020-02-11:
There is some conversation going on the **Draft Pull Request** that I created in Flathub; it would be nice if some developer(s) can clear that out to see if there is anything that can be done to get this going, otherwise we just call it a day 😎

lurch, 2020-02-11:
AFAICT [this](https://github.com/flathub/flathub/pull/1363/#issuecomment-584217903) is the conversation that @x80486 is referring to :stuck_out_tongue_winking_eye:

cherryland, 2020-02-11:
> AppImage
> Download from random website
> double-click to open
this is windows-tier user experience, absolute cancer

cherryland, 2020-02-11:
> be me
> stumble over AppImage file
> Google what this is and how to use is
> ok apparently you have to download *.AppImage files from random websites
> and double-click to use

this is shit-tier user experience, absolute windows cancer confirmed

thundron, 2020-02-12:
Thanks @lurch !
I gave an answer there, but to recap - this seems to be blocked with the current state of flatpak. Closing, hopefully there will be a chance to do it in the future with more flexibility

closed 2020-02-12 by thundron

SecretAgentAgentX, 2020-12-25:
> Do the current AppImages not work properly for you?
> We're currently using [electron-builder](https://github.com/electron-userland/electron-builder) to create our Etcher packages (including [...]

Flatpaks have worked on every distro I've used since LONG before your post. BOTH arch based and Ubuntu based.

Danik1601, 2022-05-04:
> We're currently using [electron-builder](https://github.com/electron-userland/electron-builder) to create our Etcher packages (including AppImages, debs and rpms), and that doesn't [yet](https://github.com/electron-userland/electron-builder/issues/512) support creating flatpaks.

Seems like [something happening](https://github.com/electron-userland/electron-builder/pull/5711) in [that issue](https://github.com/electron-userland/electron-builder/issues/512)

Danik1601, 2022-05-30:
> Seems like [something happening](https://github.com/electron-userland/electron-builder/pull/5711) with [that issue](https://github.com/electron-userland/electron-builder/issues/512)

As I mentioned in [my other comment](https://github.com/electron-userland/electron-builder/issues/512#issuecomment-1141444837), someone even [made](https://github.com/electron-userland/electron-builder/pull/6893) the documentation now. An it was added in the [latest pre-release](https://github.com/electron-userland/electron-builder/releases/tag/v23.1.0)

mcraa, 2022-05-31:
We are migrating the CI of etcher to gh actions, so it will be easier for contributors (and for us) to add new outputs.

Danik1601, 2022-05-31:
> We are migrating the CI of etcher to gh actions, so it will be easier for contributors (and for us) to add new outputs.

I completely do not understand where should I open the issue then

mcraa, 2022-06-01:
No worries, this is fine for now.

If it will be that simple 
someone may open directly a PR to extend our gh actions with flatpak 
after we start using gh actions.

lurch, 2022-06-04:
> We are migrating the CI of etcher to gh actions

That's great to hear, as it'll allow non-Balena-people to actually see the reason for any PR build failures :slightly_smiling_face: