balena-io/etcher #2020: The Writer Process Ended Unexpectedly - Etcher

cavalier666-uk, 2018-02-06:
get error - the writer process ended unexpectedly, when clicking write, windows 10 and ntfs 8gb pen drive, also same when run as administrator

tried latest version installed and portable, both do the same

windows 10 X64
recalbox.img - file
[image]

- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

jhermsmeier, 2018-02-06:
@cavalier666-uk is there any place we can obtain the image in question? Would like to give it a run & see if we can reproduce this.

lurch, 2018-02-07:
@jhermsmeier I'd guess it's one of these? https://archive.recalbox.com/

cavalier666-uk, 2018-02-07:
Hi yes that's the image. The pc x64 version

l4gu5, 2018-02-08:
I had the same issue regarding NOOBS and Raspbian stretch image.
I removed the new version of etcher and installed the old 1.2.1 version and it works fine again, I hope that helps. My system is a Windows 10 64 bit

cavalier666-uk, 2018-02-08:
thanks @l4gu5 will give that a go see if it works

cavalier666-uk, 2018-02-08:
thanks again, confirmed working fine on 1.2.1

lurch, 2018-02-08:
Labelling this as a bug, as it appears to be a regression from v1.2.1

emon0509, 2018-03-19:
Please help me to find the 1.2.1 version or explain how to fix this issue

lurch, 2018-03-20:
Older releases of Etcher can be found at https://github.com/resin-io/etcher/releases

jhermsmeier, 2018-04-06:
@l4gu5 NOOBS cannot be flashed with Etcher, NOOBS is not a disk image, and needs a different installation procedure as described in https://www.raspberrypi.org/documentation/installation/noobs.md

closed 2018-10-15 by jviotti

famaia, 2019-05-25:
I also had this problem today burning the latest version of RETROPIE (retropie-4.4-rpi2_rpi3.img) with the latest version of Etcher-Portable-1.5.39-x64 on Windows 10 x64.

After downloading Etcher 1.2.1, I was able to write the IMG successfully.

thundron, 2019-05-26:
@famaia closed in favor of this one https://github.com/balena-io/etcher/issues/2757 please write there the output you see in Etcher's console (see last comment)

splatkin, 2019-05-28:
right-click zip file > properties > check "unblock". Worked for me after that

MaXiNom1, 2019-07-22:
Getting the same error vr 1.5.51 mac OSX HighSierra

prajwalkoirala23, 2019-08-07:
same for me homeboy

daton89, 2019-08-08:
Getting this error on Windows10 with 1.5.53

Robert923, 2019-08-22:
Just had this issue - it was the Windows 10 Ransomware Protection.
Windows Security | Virus & threat protection | Ransomware protection (at the bottom)
Then you can either:
- temporarily disable Controlled folder access (easier) or
- click on "Allow an app through Controlled folder access", and then choose "Add an allowed app".  I find choosing the recently blocked apps is easiest.  Then just select balena-etcher (NOT the installer, that won't help).

I hope that helps,

Robert

you can

DominoX, 2019-08-23:
Etcher 1.5.6 is still giving this error first reported here last year. Win 10 Pro with latest updates (10.0.18362 Build 18362)

1.2.1 still works

The 'Ransomware protection' work-around listed by @Robert923 did not work for me. Neither of the 2 methods shown

CDeLeon94, 2019-10-01:
https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047
> ```shell
> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> ```
This worked for me

ReubenBTalbott, 2019-10-01:
> [#2833 (comment)](https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047)
> 
> > ```shell
> > sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> > ```
> 
> This worked for me
> 
> 1.5.57
> macOS Catalina Beta (19A573a)

That won't work on windows since windows does not have sudo, and windows stores apps in a different location than a Mac.

CDeLeon94, 2019-10-02:
"Run as administrator"?

ChrisRiddell, 2019-10-03:
> [#2833 (comment)](https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047)
> 
> > ```shell
> > sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> > ```
> 
> This worked for me
> 
> 1.5.57
> macOS Catalina Beta (19A573a)

Thanks that worked for me on macOS Catalina Beta  (19A578c)

lcrostarosa, 2019-10-19:
Had same issue on catalina prod this command worked for me as well 

```
    sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher

```

lurch, 2019-10-19:
@lcrostarosa For problems with Catalina, see #2833

AlexandruBuhai, 2019-10-23:
> [#2833 (comment)](https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047)
> 
> > ```shell
> > sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
> > ```
> 
> This worked for me
> 
> 1.5.57
> macOS Catalina Beta (19A573a)

Thanks! This was helpful!

Borewit, 2019-12-23:
I had the same issue on Windows 10.
Downgraded from [v1.5.70](https://github.com/balena-io/etcher/releases/tag/v1.5.70) to [v1.2.1](https://github.com/balena-io/etcher/releases/tag/v1.2.1) which allowed me to flash.

thundron, 2019-12-24:
I'd argue it's the same *permission* issue if it's a different OS, as permissions on Windows are *very* different than OSX/Linux.
You could consider looking for other Windows permission issues, or opening a new one if you don't find any similar to what you experienced

Priyansu67, 2020-03-15:
I had the same issue and it was the controlled folder access which was blocking the etcher from acessing the image file. Just turn it of in settings will do the charm.

sushah, 2020-03-19:
I had the same issue with portable version 1.5.79. Tried native windows installation - resulting in same error. Tried other versions, all with different errors. In the end, burned image with rufus and this worked without any issues.

EmnaX, 2020-03-23:
I also ran into the same issue with 1.5.79. I'm running on macOS Catalina 10.15.3 (19D76). Running BalenaEtcher with sudo worked.

lurch, 2020-03-23:
@EmnaX Your feedback may be useful on #3107 ?

faaps, 2020-04-12:
> Just had this issue - it was the Windows 10 Ransomware Protection.
> Windows Security | Virus & threat protection | Ransomware protection (at the bottom)
> Then you can either:
>
> * temporarily disable [...]

Thanks a lot!, you save me!

IIWare, 2020-08-19:
quit the app and reopen it, if it doesn't work then quit the app and run it as administrator, and I am sure I am the only one who wrote /issues/2020 lol

longletian, 2020-09-29:
[image]
 system version   win10 2004,  I've already used the administrator

thundron, 2020-09-29:
@longletian please update to the latest version and retry. If it still doesn't work, check for an already open issue, this one is quite polluted with many other issues of old versions

longletian, 2020-09-30:
thanks，The answer has been found

stijojoseph, 2020-10-16:
> Just had this issue - it was the Windows 10 Ransomware Protection.
> Windows Security | Virus & threat protection | Ransomware protection (at the bottom)
> Then you can either:
>
> * temporarily disable [...]

u saved my 125mb etched download LOL thanks a lot