balena-io/etcher #2024: Eliminating popup asking to format disk on etcher

aamm132, 2018-02-07:
- **Etcher version: 1.3.1** 
- **Operating system and architecture: Windows** 

I just recently built etcher locally and can make my changes I need to on my machine. One thing I have been looking to want to change is the popup that appears frequently whenever an image is being flashed to a drive. Is there any possibility that this popup can be prevented?

lurch, 2018-02-08:
This popup is coming from Windows itself (it's getting "upset" that it can't read any filesystems on the drive), so no there's nothing we can do (currently) to prevent it.

Longer-term, we plan to do native drive-locking, so that Etcher will be able to grab exclusive access to the drive, and (I think) Windows wouldn't then ask you to format it. However this relies on a feature that's only recently been added to NodeJS, and we need to wait for that version of NodeJS to be incorporated into Electron, before we can then add this drive-locking functionality to Etcher.

closed 2018-02-12 by aamm132

deng1604, 2020-11-09:
[image]
该如何应对?

CustodialMatter, 2026-11-09:
I'm still getting the windows format pop-ups, a fix wasn't possible?