balena-io/etcher #2028: Hangs at startup when writing to USB on mac

madurst08, 2018-02-08:
- **Etcher version:1.3.1** 
- **Operating system and architecture: Mac OSx: 10.12.6 (16G1114) 64bit** 
- **Image flashed: ubuntu-16.04.3-desktop-amd64 ** 
- **Do you see any meaningful error information in the DevTools? 
No- Full devtools console is here:

/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk scanner:init +0ms Scanner
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk subscribe +2ms StandardAdapter
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk subscribe +1ms USBBootAdapter
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.3.1, Type = dmg
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:30:45 GMT-0600 (CST) Application start ({"packageType":"dmg","version":"1.3.1"})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:30:45 GMT-0600 (CST) Checking for updates ({"currentVersion":"1.3.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk start +431ms true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:30:57 GMT-0600 (CST) Update notification skipped ({"reason":"Latest version"})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:31:24 GMT-0600 (CST) Open image selector
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:31:44 GMT-0600 (CST) Select image ({"path":"/Users/Matthew/Downloads/ubuntu-16.04.3-desktop-amd64.iso","extension":"iso","size":{"original":1587609600,"final":{"estimation":false,"value":1587609600}},"hasMBR":true,"hasGPT":true,"partitions":[{"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","id":"4047758F-244E-4075-9261-D188046DC76D","name":"ISOHybrid","firstLBA":0,"lastLBA":3100743,"extended":false},{"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","id":"79BB24B5-244E-4024-9260-E188046DDCE5","name":"ISOHybrid1","firstLBA":3006684,"lastLBA":3011355,"extended":false}],"logo":false,"bmap":false})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk stop +1m true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:31:56 GMT-0600 (CST) Progress (undefined): 0% at 0 MB/s (eta undefineds)
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:31:56 GMT-0600 (CST) Flash ({"image":"/Users/Matthew/Downloads/ubuntu-16.04.3-desktop-amd64.iso","drive":{"size":2097152000000,"isVirtual":null,"enumerator":"diskutil","logicalBlockSize":null,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":null,"adapter":"standard","isSCSI":null,"isRemovable":null,"device":"/dev/disk2","busVersion":"0.0","isSystem":false,"busType":"UNKNOWN","isCard":null,"isUSB":null,"mountpoints":[{"path":"/Volumes/U"}],"description":"UDisk","isUAS":null},"uuid":"653d6026-00ce-4adc-8740-2114f482afa2","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Elevating command: /Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper /Applications/Etcher.app/Contents/Resources/app.asar
2/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 08 2018 12:31:56 GMT-0600 (CST) Progress (undefined): 0% at 0 MB/s (eta undefineds)
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Successfully connected to IPC server: etcher-server-17390, socket root /var/folders/m4/9xcgyxjd7l39vfq7m92g60280000gr/T/
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Image: /Users/Matthew/Downloads/ubuntu-16.04.3-desktop-amd64.iso
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Device: /dev/disk2
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Umount on success: true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Validate on success: true** 

Utility monitor says its still trying
[image]

This is the screen it's stuck on:
[image]

jhermsmeier, 2018-02-12:
I see there's the Etcher Password Prompt still running - did you put in your credentials, or could the problem be that that window lost focus and is "hidden" behind Etcher?

madurst08, 2018-02-15:
I was able to find that password screen that got sent behind all my windows, entered in my user password- and is still stuck at the same spot. Trying a new new flash drive and a different iso. 

Is there a way pull the password prompt up or just run etcher in root?

Devtools output:
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk scanner:init +0ms Scanner
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk subscribe +292ms StandardAdapter
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk subscribe +2ms USBBootAdapter
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@resin.io

Version = 1.3.1, Type = dmg
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:03:29 GMT-0600 (CST) Application start ({"packageType":"dmg","version":"1.3.1"})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:03:30 GMT-0600 (CST) Checking for updates ({"currentVersion":"1.3.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk start +15s true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:03:55 GMT-0600 (CST) Open image selector
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:04:10 GMT-0600 (CST) Update notification skipped ({"reason":"Latest version"})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:04:54 GMT-0600 (CST) Select image ({"path":"/Users/Matthew/Downloads/ubuntu-16.04.3-desktop-i386.iso","extension":"iso","size":{"original":1607467008,"final":{"estimation":false,"value":1607467008}},"hasMBR":true,"hasGPT":false,"partitions":[{"type":23,"id":null,"name":null,"firstLBA":64,"lastLBA":3139584,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false},{"type":0,"id":null,"name":null,"firstLBA":0,"lastLBA":0,"extended":false}],"logo":false,"bmap":false})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk stop +2m true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:05:05 GMT-0600 (CST) Progress (undefined): 0% at 0 MB/s (eta undefineds)
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:05:05 GMT-0600 (CST) Flash ({"image":"/Users/Matthew/Downloads/ubuntu-16.04.3-desktop-i386.iso","drive":{"size":15623782400,"isVirtual":null,"enumerator":"diskutil","logicalBlockSize":null,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":null,"adapter":"standard","isSCSI":null,"isRemovable":null,"device":"/dev/disk2","busVersion":"0.0","isSystem":false,"busType":"UNKNOWN","isCard":null,"isUSB":null,"mountpoints":[{"path":"/Volumes/U"}],"description":"STORE N GO","isUAS":null},"uuid":"a7ce876a-c05c-426e-9323-b688b9a76452","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Feb 15 2018 15:05:05 GMT-0600 (CST) Progress (undefined): 0% at 0 MB/s (eta undefineds)
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Elevating command: /Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper /Applications/Etcher.app/Contents/Resources/app.asar
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Successfully connected to IPC server: etcher-server-41129, socket root /var/folders/m4/9xcgyxjd7l39vfq7m92g60280000gr/T/
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Image: /Users/Matthew/Downloads/ubuntu-16.04.3-desktop-i386.iso
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Device: /dev/disk2
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Umount on success: true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Validate on success: true

kenorb, 2018-05-13:
I had _Etcher_ stuck at _Starting..._ step as well, but it was `diskutil` issue.

I've checked `pstree` and it was running `sudo diskutil info -plist /dev/disk3`:

```
 | | \-+- 23600 kenorb ./applet
 | |   \-+- 23616 root /bin/bash ./sudo-prompt-script
 | |     \-+- 23620 root /bin/bash sudo-prompt-command
 | |       \-+- 23621 root /Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper /Applications/Etcher.app/Contents/Resources/app.asar
 | |         \--- 23633 root diskutil info -plist /dev/disk3
```

When I run this manually, it got stuck as well:

```
$ sudo diskutil info -plist /dev/disk3

load: 1.68  cmd: diskutil 31161 uninterruptible 0.01u 0.01s
load: 1.68  cmd: diskutil 31161 uninterruptible 0.01u 0.01s
load: 2.02  cmd: diskutil 31161 uninterruptible 0.01u 0.01s
load: 2.02  cmd: diskutil 31161 uninterruptible 0.01u 0.01s
^C^C^C^C
^C^C^C
```

but it worked when I run without `sudo`:

```
$ diskutil info -plist /dev/disk3
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Bootable</key>
	<false/>
	<key>BusProtocol</key>
	<string>USB</string>
	<key>CanBeMadeBootable</key>
	<false/>
	<key>CanBeMadeBootableRequiresDestroy</key>
	<false/>
	<key>Content</key>
	<string>GUID_partition_scheme</string>
	<key>DeviceBlockSize</key>
	<integer>512</integer>
...
```

I've generated spindump for `diskutil` and I've reported this [bug to Apple](https://bugreport.apple.com/).

closed 2018-10-15 by jviotti