balena-io/etcher #2038: ETA is extremely unrealiable

FredeEB, 2018-02-12:
- **Etcher version: 1.3.1** 
- **Operating system and architecture: Windows 10, 1709, x64** 
- **Image flashed: Raspian** 
- **Do you see any meaningful error information in the DevTools? N/A** 

The ETA is fluctuating up and down by about 20 minutes at short regular intervals (about 1 second), making it completely useless. A concrete example I've had is between 4:32 and 25:54 minutes 'til finished. I haven't dug around in the code, so I wouldn't know the cause. It seemed like there are two independent times, because when the low one, went past zero, it is now back at about 50 minutes, and the, before, high one is still counting down.

Not a huge problem for me, but definitely an inconvenience, and I could see it being a bigger issue at larger scale.

lurch, 2018-02-15:
I wonder if this might be because the write-speeds on your drive are also fluctuating? What USB-drive / SD-card are you writing to? (and if it's an SD card, what card-reader are you using?)

FredeEB, 2018-02-15:
Definitely a possibility, as I'm using generic 8 GB micro SD cards in a crappy generic cardreader, Might have to get a new one and check if it solved the problem

jhermsmeier, 2018-02-15:
This could be caused by writes intermittently failing, as the write speed would naturally drop while it's retrying writes. So, I agree with what @lurch suggested, that it's more likely the culprit is the card reader or SD-card itself. Let us know how it goes with a different setup, and if the problem persists.

FredeEB, 2018-02-15:
I will!

lurch, 2018-02-15:
@jhermsmeier Might it make sense to have the speed that Etcher displays (and thus the ETA that it calculates) be a "rolling average" speed, rather than an "instantaneous" speed?

FredeEB, 2018-02-15:
@lurch Might be a solution for more minor fluctuations, It would be some complex math, seeing as you'd have to calculate an average but still subtract time from it. In my case it would result in a completely arbitrary value though, as the ETA shifted really far. It does seem hardware related though, So it might just be a problem the people buying cheap cardreaders (like me) would have to deal with.

jhermsmeier, 2018-02-15:
@lurch it is a rolling average over several seconds, but the write retries follow a backoff that will still cause the write speed to drop – and besides, why would we want to intentionally display an incorrect write speed?

FredeEB, 2018-02-15:
@jhermsmeier Makes sense. I'll get my hands on a new cardreader and return

FredeEB, 2018-02-17:
@jhermsmeier It works perfectily with a new cardreader,

Maybe the solution would be to check for large fluctuations in writespeed and inform the user about the ETA being unreliable, or maybe just disable it outright because of it?

lurch, 2018-02-18:
@FredeEB I think you might be the first person to report this problem, so I'm not sure if it'd be worth adding special-case code to Etcher to try and "detect" fluctuations in writespeed?

closed 2018-05-08 by FredeEB

Langhalsdino, 2019-08-14:
I got a similar problem while bruning 512GB images on 512GB SD cards. The reported ETA is displayed to be about 30 to 40 min but it takes around 7h to complete. Furthermore the ETA seems to be following some kind of exponential decay. 

It would be awesome if the written file size can be displayed e.g 289GB of 512GB has been written to the device. Its a better progress estimation then the wrong percent and time estimates ;)
The main reason, why the burned percent is off might be that i use a .zip img as source file.

Furthermore i am burning the image to about 6 SD Cards simultaneously and therefore there might be some fluctuations in writespeed.

lurch, 2019-08-14:
@Langhalsdino How big is the original uncompressed image? How big is the compressed zip file? Do you get more "reliable" behaviour if you flash the uncompressed image rather than the compressed zip? See #2851