balena-io/etcher #2041: Unable to flash a new SD card

bluan, 2018-02-14:
Etcher failed to flash the AIY image to a new Sumsung 128GB SDXC card:
- **Etcher version:** 1.3.1
- **Operating system and architecture:**  Windows 10 Enterprise, 64bit version
- **Image flashed:** aiyprojects-2018-01-03.img.xz
- **Do you see any meaningful error information in the DevTools?** 
Error Message:  Access to the specified file is denied

In the DevTool Console, message is the following:
({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
C:\Users\bluan\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 
{"name":"Error","message":"Access to the specified file is denied","cause":
{},"isOperational":true,"stack":"Error: Access to the specified file is denied"}
console.(anonymous function)  
@ C:\Users\bluan\AppData\Local\Programs\etcher\resources\app.asar\node_modules\resin-corvus\src\resin…:228
exports.report 
.....

lurch, 2018-02-14:
Is your `aiyprojects-2018-01-03.img.xz` stored on a network-share, e.g. a NAS?
Sometimes there's an issue where the image file on a network-share is readable (and therefore selectable) by the 'normal' user that Etcher runs as, but when Etcher then elevates to an Admin user (in order to have the necessary permissions to perform low-level disk writes) the Admin user isn't authorised to read from the network share.
If that _is_ the case, then the current workaround is to just copy the image file to a local hard-drive before attempting to flash it with Etcher.
If that's _not_ the case, then this is obviously a different bug so ignore everything I wrote above ;-)

bluan, 2018-02-14:
Thanks a lot for your reply and comments!  The image file is on a local harddisk where the Etcher tool is also located, so yeah, it's a different bug :-) 

Your comments makes me to wonder one thing though... on my laptop, which is a company machine, has a Bit9 program running, which is a corprate security software the IT department puts on our machines to prevent any software not whitelisted from running. So I wonder if the Ether tool also invoke any other program when it flash the SD, which could be blocked by Bit9.  But the error message certainly does not say that, other than "denied", so exactly what's being "denied" and by who, I wish the error logging by Etcher could say more. Maybe more robust logging is a feature they could add in the next release....

lurch, 2018-02-14:
Thanks for the extra info @bluan 
Could you clarify exactly **when** you see the error message? Before selecting the image / after selecting the image / when you first press the Flash button / after the Flash process has already started? etc.

bluan, 2018-02-14:
Yes, the error displays after clicking on the Flash! button - see attached image file below:
[image]

Here are more error messages from the console:

Wed Feb 14 2018 10:50:35 GMT-0800 (Pacific Standard Time) Application start 
({"packageType":"nsis","version":"1.3.1"})
C:\Users\bluan\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Wed Feb 14 2018 10:50:35 GMT-0800 (Pacific Standard Time) 

Checking for updates ({"currentVersion":"1.3.1","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})

C:\Users\bluan\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 
sdk start +452ms true

api.mixpanel.com/track/?data=eyJldmVudCI6ICJtcF9wYWdlX3ZpZXciLCJwcm9wZXJ0aW…uIjogIjYzZTVmYzQ1NjNlMDA5MjhkYTY3ZDEyMjYzNjRkZDRjIn19&ip=1&_=1518634235824 

Failed to load resource: net::ERR_INTERNET_DISCONNECTED

api.mixpanel.com/track/?data=eyJldmVudCI6ICJDaGVja2luZyBmb3IgdXBkYXRlcyIsIn…lMDA5MjhkYTY3ZDEyMjYzNjRkZDRjIiwiJF9fYyI6IDB9fQ%3D%3D&ip=1&_=1518634235831 

Failed to load resource: net::ERR_INTERNET_DISCONNECTED

api.mixpanel.com/track/?data=eyJldmVudCI6ICJBcHBsaWNhdGlvbiBzdGFydCIsInByb3…jNDU2M2UwMDkyOGRhNjdkMTIyNjM2NGRkNGMiLCIkX19jIjogMH19&ip=1&_=1518634235827 

Failed to load resource: net::ERR_INTERNET_DISCONNECTED

C:\Users\bluan\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 {"name":"Error","message":"getaddrinfo ENOENT resin-production-

downloads.s3.amazonaws.com:443","cause":{"code":"ENOENT","errno":"ENOENT","syscall":"getaddrinfo","hostname":"resin-production-downloads.s3.amazonaws.com","host":"resin-production-downloads.s3.amazonaws.com","port":443},"isOperational":true,"stack":"Error: getaddrinfo ENOENT resin-production-downloads.s3.amazonaws.com:443\n    at Object.exports._errnoException (util.js:1050:11)\n    at errnoException (dns.js:33:15)\n    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:73:26)","code":"ENOENT","errno":"ENOENT","syscall":"getaddrinfo","hostname":"resin-production-downloads.s3.amazonaws.com","host":"resin-production-downloads.s3.amazonaws.com","port":443}

console.(anonymous function) @ C:\Users\bluan\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55
C:\Users\bluan\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55

bluan, 2018-02-14:
BTW, the Internet disconnected error is wrong!  - I am sitting in my office with a fast and reliable Internet connection at all time.

bluan, 2018-02-14:
Hi, the problem has been fixed now, after I went to talking to the corp tech support folks, and I was told that the location settings in my laptop might be the cause of the problem. After I reset it, it now flashed without problem.  Thank you all!

lurch, 2018-02-15:
Oh, how weird! Glad to hear that it's all working now though :-)
Can this issue be closed?
(or do you want to try digging into this a bit further @jhermsmeier ?)

bluan, 2018-02-15:
We can close the issue, but let's keep the log here if possible, as the AIY engineers may want to take a look at the logs...  Thanks!

lurch, 2018-02-15:
Yup, closing an issue leaves the comments publicly viewable :)

closed 2018-02-15 by lurch

abhibhatia98, 2019-08-19:
Hi
when i am flashing the image in sd card its shows error   **" something went wrong"** please help.
I am using windows 10 (64).

abhibhatia98, 2019-08-19:
Hi
when i am flashing the image in sd card its shows error " something went wrong" please help.
I am using windows 10 (64). I am sharing screenshot as well. 
[image]

lurch, 2019-08-19:
@abhibhatia98 You might want to add your comments to #2854 instead of commenting on a closed issue.