balena-io/etcher #2046: ArchLinux32, Etcher i686

Jakecheslekov, 2018-02-14:
- **Etcher version:** N/A
- **Operating system and architecture:** Archlinux32.org i686 
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

Hey guys, so an unofficial 32-bit port of ArchLinux has sprouted up at archlinux32.org, I was wondering if you guys control the AUR entry for Etcher or not? It would seem that it no longer supports 32-bit, about a month or two ago the unofficial archlinux32.org sprouted up & is now running great, at least in my opinion. I was wondering if you guys could make an Etcher AUR entry for i686 arch.

jhermsmeier, 2018-02-14:
I'm pretty sure we had a similar issue on here before, but can't find it anymore...
Either way – we currently do not have control over the AUR packages for Etcher, and I'm not sure who it was who's making them. We were recently discussing this topic though, and thinking about providing more packages directly.

Jakecheslekov, 2018-02-14:
Is there a way to install Etcher-gui with NPM? Or is it etcher-cli only that way? 
Thank you for the speedy labels Jhermsmeier, I love this program & I love your speedy support even more!
It would be nice if there were a way to build it from the git repo, like a makefile or something that would work universally on all Linux distros, excuse my ignorance if this is already possible but last I checked I don't think it was. Perhaps it could be added to 'pip' the python-pip thing? 

I've never made an AUR entry before but I would gladly give it a shot! Maybe I could just copy the PKGBUILD from the other one?
And yeah I don't know what happened to that first "issue" but I remember seeing it a while ago... Odd how things sometimes just vanish.

As far as nonfree stuff goes, is Etcher 100% free software? I'd like to get Etcher added to Parabola & Archlinux32.

lurch, 2018-02-15:
Looking through my emails, it seems the most recent AUR discussion was in #2007 , but it looks like the person who opened that issue ( @kenny-wettlaufer ) actually deleted it (I didn't even know you could do that!) rather than just closing it.

lurch, 2018-02-15:
Hmmm, or maybe Kenny deleted his account, which also automatically deleted all the Etcher issues he created? :-/ ( #1964 , #1999 and #2007 )

jhermsmeier, 2018-02-15:
Oh wow, didn't know that deleting your account makes everything disappear :/

lurch, 2018-02-28:
> It would be nice if there were a way to build it from the git repo, like a makefile or something that would work universally on all Linux distros

@Jakecheslekov See https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md (although note that there's currently a step missing #2079 )

> As far as nonfree stuff goes, is Etcher 100% free software?

https://github.com/resin-io/etcher/blob/master/LICENSE