balena-io/etcher #2053: Etcher crashed, my MicroSD is now corrupt and even Diskpart won't fix it

BaconCatBug, 2018-02-15:
Etcher crashed mid flash, and now I can't get the SD card to work at all. Etcher still detects the card and the capacity, but neither diskpart nor EaseUS partition manager can fix it. 

Diskpart now reports a System I/O error and trying to flash it also fails http://prntscr.com/if94d2

Has etcher permanently damaged my card? It was working fine, I'd flashed a different linux OS onto it before, and was putting a new one on when it crashed.

Tonymac32, 2018-02-19:
Did you try SDFormatter, option "format size adjustment?"

Just in case, I've had a lot of tools fail to properly identify a mangled card.

leddmail, 2018-02-21:
Hello guys, I've joined GitHub just to say thanks! I've had the same issue and this was the only way I could fix it!

lurch, 2018-02-21:
@jhermsmeier Do you think #2023 will fix this problem?

jhermsmeier, 2018-02-21:
@lurch depends; the primary issue described here – that the card can't be flashed / formatted anymore on Windows – is that Windows can't deal with partition tables properly. If the partition table doesn't map the entire storage device, or doesn't fit what's actually on the storage device, Windows goes nuts. This has been known for a while, and there are even partition tables that can make the Blue Screen Of Death happen.

The resulting problem is that diskpart blows – and that's what #2023 should solve, and so make flashing those storage devices with mangled partition tables with Etcher possible again.

The former we can't fix, the latter we hopefully will have in the next version.

lurch, 2018-02-23:
@BaconCatBug or @Tonymac32 or @leddmail Could you please try this version of Etcher, and let us know if that fixes the problem for you? https://resin-nightly-downloads.s3.amazonaws.com/etcher/2018-02-20/1.3.1%2B5be64e1/Etcher-Portable-1.3.1%2B5be64e1-x64.exe

Tonymac32, 2018-02-25:
I'm afraid I've never seen the problem, so I have nothing to compare with.  I'm still using 1.0.0-rc.4

lurch, 2018-02-25:
> I'm still using 1.0.0-rc.4

Wow, that's a really old version - there's been a [number of bugfixes and improvements](https://github.com/resin-io/etcher/blob/master/CHANGELOG.md) since then. Just out of curiosity, why are you still using that version?

Tonymac32, 2018-02-25:
Laziness, and lack of issues.  I didn't actually know what version I had until I checked because of this thread.  I'm just lucky I guess?

jhermsmeier, 2018-03-26:
Closing due to lack of response; @BaconCatBug feel free to reopen if you get back around to it

closed 2018-03-26 by jhermsmeier