balena-io/etcher #2057: etcher spies on the user without consent

sneak, 2018-02-18:
etcher version 1.3.1

This is the part of #1718 that is not yet resolved.

I launch the app and flash a disk, and it transmits my activity to multiple third parties (google analytics and mixpanel and some other one) without my consent or even warning me that it's doing so so that I could e.g. discontinue use of the spyware app.

Please fix this.  Transmitting user data without ADVANCE, OPT-IN consent is unethical.

jviotti, 2018-02-19:
I double-checked this and can confirm no requests to any service (we don't even use Google Analytics, as claimed here) is made from the application. Closing as invalid. Please re-open if you can point us to an "unethical" request.

[image]

closed 2018-02-19 by jviotti

kevcampb, 2018-11-29:
I started up etcher (1.4.8), and it certainly makes a large number of tracking requests on startup, without any opt-in to analytics.

[image]

Content is now sent over HTTPs, so I don't have the internals of these. Google analytics is certainly included. The previous version of etcher certainly included tracking codes in the messages sent to services.

There is a setting "Anonymously report errors and usage statistics to balena.io" in the settings, but it is defaulted to on.