balena-io/etcher #2069: Implement average progress with multi-writes

Shou, 2018-02-21:
There are multiple things we need to consider:

* Transitioning from flashing to verifying
  - Once writing finishes, show average flash that has been happening in the background
* Keeping the average consistent
  - Solution: cache devices that are done and consider them at 100%
  - Ensures the progress doesn't jump backwards

jhermsmeier, 2018-02-21:
@Shou is this specified somewhere? If so, can you add links to it?

Shou, 2018-02-21:
@jhermsmeier it was briefly discussed in the product call today, whether to use the slowest drive or an average of all – the average was selected, so this is just a list of things to consider when implementing it.

jhermsmeier, 2018-03-23:
Resolved by #2124

closed 2018-03-23 by jhermsmeier