balena-io/etcher #2071: Unsafe Drives Displayed/Selectable

turbouser, 2018-02-21:
- Etcher version: Etcher-Portable-1.3.1-x64.exe
- Operating system and architecture: Windows 8.1 x64
- Image flashed: 2017-11-29-raspbian-stretch-lite.img
- Do you see any meaningful error information in the DevTools? No

Unsafe mode is NOT checked, but the following drives are being offered in addition to the expected flash drive:

WD My Book 1230 USB Device - 4 TB
WDC WD1600ADFS75SLR2 - 160 GB

These drives have never been shown in any version of Etcher prior to 1.3.1 (unless Unsafe mode was checked).

jhermsmeier, 2018-02-21:
This is because the drive detection has undergone a significant change in 1.3.1, which now eliminates devices based on how they're connected. I'm guessing those two drives are connected via USB?

turbouser, 2018-02-21:
WD My Book is an external 4 TB USB drive used for daily system backups

WDC WD1600ADFS75SLR2 is an internal SATA system drive.

Allowing Etcher to write to either of these drives would be a disaster here.  I would prefer they both remain hidden as they have been in all previous versions of Etcher.

lurch, 2018-02-22:
Just FYI the drive-detection component that's recently had a big change is https://github.com/resin-io-modules/drivelist
So if you'd like to install that and let us know what it outputs for your drives, that'd be really helpful.

Also, in the next version of Etcher there'll be a feature ( #2045 ) that'll warn you about any drives bigger than 64GB.

turbouser, 2018-02-22:
I don't see an executable version of drivelist for Windows 8.1 x64.

lurch, 2018-02-23:
Pinging @jhermsmeier for drivelist-on-Windows help :)

turbouser, 2018-02-28:
Any chance you could add a user option to hide unsafe drives that are showing up so there's absolutely no chance of accidentally selecting one of them?

lurch, 2018-02-28:
You mean like a blacklist? We previously discussed that in #1728 which details the complexities of such an approach.

jhermsmeier, 2018-02-28:
Etcher can only do so much; and this generally is a thing where you need to pay attention to what you're doing – if it's displaying devices that should be deemed "unsafe", then that's a bug – but I don't think putting in safety-cages at every turn is a good idea. It hurts usability considerably, and in the end you still need to be aware of what's going on.

lurch, 2018-02-28:
@jhermsmeier the "internal SATA system drive" shouldn't be being displayed if unsafe mode is disabled, right?

turbouser, 2018-03-05:
>You mean like a blacklist? We previously discussed that in #1728 which details the complexities of such an approach.

Is it really that complex to have a "Settings" option that displays a list of drives that would normally be presented by "Select drive"?  Each drive in that "Settings" option list would simply have a checkbox, that if checked, would hide that drive from the list presented by "Select drive".

jhermsmeier, 2018-03-21:
@lurch yup, that's correct – internal storage devices should not make the list, unless "Unsafe mode" is enabled; @turbouser can you confirm that the internal WDC WD1600ADFS75SLR2 - 160 GB shows even when unsafe mode is disabled?

turbouser, 2018-03-21:
>turbouser can you confirm that the internal WDC WD1600ADFS75SLR2 - 160 GB shows even when unsafe mode is disabled?

As stated in my original post: Unsafe mode is NOT checked.  It did not show up until v1.3.1.

jhermsmeier, 2018-03-21:
Thanks! Was asking because I can't reproduce this so far; would you be interested in running drivelist's example & posting the output, so we can debug this?

turbouser, 2018-03-21:
I would be happy to run drivelist, but as I commented 27 days ago, it's not clear to me how to do so under Windows 8.1 x64.  Everything I've found appears to be Linux oriented.

jhermsmeier, 2018-03-21:
Aye, that's why I was asking if you'd be happy to do it at all :) I've built you two executables to try – you'll need to run them from the command line to grab the output, i.e.

```
drivelist-example-6.0.4.exe > drivelist-6.0.4.log
```

The two versions are first, the one that ships with Etcher v1.3.1, and second the current latest, for comparison.

1. drivelist-example-6.0.4.zip
2. drivelist-example-6.1.3.zip

turbouser, 2018-03-21:
Thanks!  The output is identical from both versions:

[drivelist-6.0.4.log](https://github.com/resin-io/etcher/files/1835459/drivelist-6.0.4.log)
[drivelist-6.1.3.log](https://github.com/resin-io/etcher/files/1835460/drivelist-6.1.3.log)

The WDC SATA drive is listed as removable (isRemovable: true), but it's not and is hung off of an SATA cable like the rest.  If you know where in Windows I can change this, please let me know.

If it can't be changed in Windows, this is another argument for my previous suggestion:

Have a "Settings" option that displays the list of drives that would normally be presented by "Select drive".  Each drive in that "Settings" option list would simply have a checkbox, that if checked, would hide that drive from the list presented by "Select drive".

jhermsmeier, 2018-03-21:
Thank you! This is interesting... do any of the RAID storage devices show up in Etcher as well (with / without unsafe mode)?

turbouser, 2018-03-21:
Everything is as expected except the WDC Sata drive:

[image]
[image]

jhermsmeier, 2018-03-22:
Hmm, we need to filter out the RAIDs even in unsafe mode though – even flashing those intentionally could lead to unintended effects.

turbouser, 2018-04-23:
FWIW, 1.43 is unchanged from my original post:

[image]

Fortunately, 1.2 works properly and doesn't put me in a vulnerable situation.

jhermsmeier, 2018-04-23:
@turbouser yeah, I haven't quite gotten around to fixing the internal one showing up (also couldn't reproduce this yet), but you'll get an additional warning now if you try to flash a device that's labeled "large"

closed 2018-10-15 by jviotti