balena-io/etcher #2072: No windows downloads available for 1.3.1

InuyashaITB, 2018-02-21:
- **Etcher version:** 1.3.1
- **Operating system and architecture:** Windows 10 - x64

There is no download links available for 1.3.1 on Windows.

jhermsmeier, 2018-02-21:
@InuyashaITB thanks for bringing this up so quickly; our build pipeline appears to have caught a bug and decided to meddle with our releases; I've fixed the v1.3.1 downloads now – will keep this issue open until the underlying issue is resolved.

MttArmstrong, 2018-02-22:
Having the same problem with MacOS as well, just a heads up! @jhermsmeier

jhermsmeier, 2018-02-22:
@MttArmstrong everything is back to normal as far as we can see; are you still having issues downloading it?

jhermsmeier, 2018-02-23:
Closing, as the underlying cause has been tended to

closed 2018-02-23 by jhermsmeier