balena-io/etcher #2073: The app creates apple partitions on a flash to boot Ubuntu

mityukov, 2018-02-22:
- **Etcher version:** 

Etcher 1.3.1

- **Operating system and architecture:** 

I use the Etcher app on MacBook (Max OS High Sierra / 10.13.3).
But the flash drive I'm going to use to install Ubuntu on a Win 10 notebook (ASUS).

- **Image flashed:** 

ubuntu-17.10.1-desktop-amd64.iso

- **Do you see any meaningful error information in the DevTools?** 

no.

After finishing flashing, I've checked partitions with `diskutil` and here's the output:
```
$ diskutil list
...
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     Apple_partition_scheme                        *8.1 GB     disk2
   1:        Apple_partition_map                         4.1 KB     disk2s1
   2:                  Apple_HFS                         2.4 MB     disk2s2
```

Any ideas on why Ubuntu installer is partitioned this way?

Here's what the flash drive was before flashing the image, by the way:
```
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *8.1 GB     disk2
   1:                       0xEE                         8.1 GB     disk2s1
```

jhermsmeier, 2018-02-22:
Hey @mityukov, Etcher flashes images as they are (copying them byte-by-byte), and so isn't creating any partitions, and doesn't do any formatting. Nevertheless the partitions on there seem odd for an Ubuntu .iso. I'll check it out.

lurch, 2018-02-22:
I suspect this is just another side-effect of the weird partitioning of Ubuntu ISOs
https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

I'm going to close this issue, but please let us know @mityukov if your USB stick isn't actually bootable.

closed 2018-02-22 by lurch

jhermsmeier, 2018-02-22:
What does `fdisk ubuntu-17.10.1-desktop-amd64.iso` and `sudo fdisk /dev/disk2` show for you?

jhermsmeier, 2018-02-22:
You're right @lurch – it's just Ubuntu's ISO Hybrid structure, which ends up confusing diskutil a little; it has all three, a Master Boot Record, GUID Partition Table and an Apple Partition Map (seen below):

```
0000000000000800  50 4d 00 00 00 00 00 02 00 00 00 01 00 00 00 02  | PM.............. |
0000000000000810  41 70 70 6c 65 00 00 00 00 00 00 00 00 00 00 00  | Apple........... |
0000000000000820  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................ |
0000000000000830  41 70 70 6c 65 5f 70 61 72 74 69 74 69 6f 6e 5f  | Apple_partition_ |
0000000000000840  6d 61 70 00 00 00 00 00 00 00 00 00 00 00 00 00  | map............. |
0000000000000850  00 00 00 00 00 00 00 02 00 00 00 03 00 00 00 00  | ................ |
0000000000000860  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................ |
 * 121
0000000000001000  50 4d 00 00 00 00 00 02 00 0a fc 07 00 00 04 80  | PM.............. |
0000000000001010  45 46 49 00 00 00 00 00 00 00 00 00 00 00 00 00  | EFI............. |
0000000000001020  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................ |
0000000000001030  41 70 70 6c 65 5f 48 46 53 00 00 00 00 00 00 00  | Apple_HFS....... |
0000000000001040  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................ |
0000000000001050  00 00 00 00 00 00 04 80 40 00 00 13 00 00 00 00  | ........@....... |
0000000000001060  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................ |
 * 121
0000000000001800  a2 a0 d0 eb e5 b9 33 44 87 c0 68 b6 b7 26 99 c7  | ......3D..h..&.. |
0000000000001810  6a 77 8a 4d f3 32 77 40 92 61 32 9d 89 a2 26 27  | jw.M.2w@.a2...&' |
0000000000001820  00 00 00 00 00 00 00 00 87 c7 2c 00 00 00 00 00  | ..........,..... |
0000000000001830  01 00 00 00 00 00 00 10 49 00 53 00 4f 00 48 00  | ........I.S.O.H. |
0000000000001840  79 00 62 00 72 00 69 00 64 00 00 00 00 00 00 00  | y.b.r.i.d....... |
0000000000001850  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................ |
```

@mityukov this is nothing to worry about; that's all expected, and should work just fine.

mityukov, 2018-02-22:
Thank you. I've just thought that Etcher for macOS would intentionally burn images able to boot on Apple hardware. And that it's just me trying to make flash drive to boot later a different machine :-)

lurch, 2018-02-22:
Nah, Etcher doesn't do anything that "clever"; all it does is copy them byte-by-byte like @jhermsmeier said (much like the command-line tool `dd` would do).