balena-io/etcher #2074: EINVAL: invalid argument, write

tangb, 2018-02-22:
- **Etcher version:** 
etcher-cli 1.3.1
- **Operating system and architecture:** 
Linux debian stable 8.10 up-to-date
- **Image flashed:** 
Raspberry pi https://downloads.raspberrypi.org/raspbian_lite_latest
- **Do you see any meaningful error information in the DevTools?** 
No DevTools on cli

I launch cli as root (not sudo) with this command:
> ./etcher --drive /dev/mmcblk0 /home/tbonneau/Downloads/2017-11-29-raspbian-stretch-lite.zip

And I got a confirmation message
> ? This will erase the selected drive. Are you sure? Yes

Finally I simply got:
> EINVAL: invalid argument, write

Is there a way to get more info ?
Is there something I can do to investigate more ?

Thank you

tangb, 2018-02-22:
I tested with older cli version and:
- 1.2.1 => failed
- 1.2.0 => success
- 1.1.2 => success

By the way, etcher application v1.3.1 works well on the same computer with the same input file.

jhermsmeier, 2018-02-22:
This is something that happens on some Linux machines; and should be fixed by #2055 in the next release of Etcher

tangb, 2018-02-22:
@jhermsmeier 
ok thank you very much for your answer

lurch, 2018-02-28:
Should this be closed then @jhermsmeier ?

closed 2018-02-28 by jhermsmeier