balena-io/etcher #2083: Slow download speed

ayrat555, 2018-02-23:
Ether installer is downloaded very slowly

Avaray, 2018-12-01:
Bumping because it's really annoying.
Download speed is about 160 kB/s... so you need to wait about 10 minutes to download Etcher.
You should make mirrors.

lurch, 2018-12-03:
Are you sure this isn't a problem with your ISP? I just downloaded it at over 3 MB/s.

Tiedemate, 2019-01-10:
Today I could download with decent speed (I had tried unsuccessfully for about a week). Can probably be closed again...

closed 2019-01-10 by ayrat555

Avaray, 2019-01-10:
In my case that was ISP. I switched to VPN (Dutch server) and I had full download speed. I have really good 100 MBit connection and I have problems only with downloading Ether. Weird. It seems my ISP don't like Etcher :P

lurch, 2019-01-10:
...or AWS?

Serjaru, 2019-10-11:
Very slow! Idiotism!! Make mirrors! idiots

lurch, 2019-10-11:
Do the links here work any better?
https://github.com/balena-io/etcher/releases

bjoernhohe, 2019-10-30:
Download speed is basically none existent - the 8 GB Image loads faster than the few MB for etcher

deelite, 2019-11-03:
Why not using torrent? My machine's ready to serve!

saqib-ahmed, 2020-07-18:
I'm having the same issue. Download speed is really slow.

thundron, 2020-07-21:
It's not something that is on our hands, the releases are distributed through GitHub, and depending on the region they might be slower/faster. Locking