balena-io/etcher #2085: "Etcher detected potential corruption issues when reading image back to the drive"

usualpoz, 2018-02-24:
- Etcher version: 1.3.1
- Operating system and architecture: Windows 7 x32
- Image flashed: kali-linux-2018.1-i386.iso https://www.kali.org/downloads/
When I try to flash this image on my USB drive it shows this message.
[image]
But when I try different USB, everything is OK.
What should I do?

lurch, 2018-02-24:
Well I guess if the first USB drive failed, but the second USB drive worked okay, then maybe the first USB drive is faulty? You can download a tool called h2testw if you want to do a more thorough test of your possibly-faulty USB drive.

GoodBlokeAri, 2018-02-24:
Happens to me with all kali versions except for 2016.3 — please help!

lurch, 2018-02-24:
@GoodBlokeAri Have you tested with different USB drives? If kali 2016.3 is the only version that can be successfully written to your USB drive, I wonder if it _might_ be because all versions after that are bigger, and so perhaps kali 2016.3 is small enough that it fits on your USB drive before the "bad area", and all later (bigger) kali versions overlap into the "bad area"?
That's just a totally wild guess though, and running the h2testw tool that I mentioned earlier will give you a much better idea of whether your USB drive is faulty or not.

GoodBlokeAri, 2018-02-24:
@lurch Hey. I'm using a 64GB USB but upon upgrading to 1.3.1 Etcher it works without error. I'm just having a problem with a nouveau modeset now lol. Thanks for your help!

GoodBlokeAri, 2018-02-26:
@lurch Now it's happening again. Same USB, same ISO. Must be my USB failing to format properly like you said.

lurch, 2018-02-26:
@GoodBlokeAri Could you clarify the exact error message you're getting please? Does it always fail at the same percentage in the verification process, or does it fail at a different point each time?
Would you also mind trying the version of Etcher I pointed to in https://github.com/resin-io/etcher/issues/2053#issuecomment-368018857 to see if that still gives you the same error?

GoodBlokeAri, 2018-02-27:
@lurch It always emits the error after validating the iso once it has been mounted. I was using that version, but it's only intermittent — I often get the error but every now and then it will work, I think it's something to do with how it formats the drive. Since UniversialUSBInstaller did it without error.

lurch, 2018-02-27:
@GoodBlokeAri Could you clarify what you mean by "emits the error after validating the iso once it has been mounted" ?
The way Etcher is _supposed_ to work, is that it writes the image to your drive, and then reads the image back from the drive, to verify that the 'output' that's now on the drive is identical to the 'input' image (with the drive remaining unmounted the whole time). But if Windows is indeed mounting the drive before (or during) verification then that could indeed screw up the verification process, as sometimes **just mounting** a filesystem is enough to modify it (e.g. some filesystems store a "last mounted" date), which would cause Etcher's verification process to detect "corruption".

> I think it's something to do with how it formats the drive

That shouldn't really matter, as Etcher works at a "low level" which means it should just overwrite whatever was previously on the drive, regardless of what the drive previously contained or how it was previously formatted. If it doesn't, then it's a bug.

> I often get the error but every now and then it will work

So, just to clarify - do both of your USB sticks behave the same, ruling out a hardware problem? Are there some images which _always_ work, and other images which sometimes work and sometimes fail? Or do _all_ images sometimes work and sometimes fail?

closed 2018-10-15 by jviotti