balena-io/etcher #2086: Mac OS X: unable to burn - destination media sizes incorrect

rinchen, 2018-02-24:
Etcher fails to recognize the available drive space on a 700mb Sony CD, as well as a DVD, and will not let me burn a 304.1MB ISO.  I was able to use the build-in Mac OS burn command just fine.

The CD and DVD were mounted in a superdrive. 

- **Etcher version:** 

1.3.1  and  1.2.1

1.3.1 says insufficient space
1.2.1 says "drive locked" (and it's a unused CD/DVD so it can't be locked)

- **Operating system and architecture:** 

Mac OS 10.13.3  iMac

- **Image flashed:** 

debian-9.3.0-amd64-netinst.iso (304.1MB)

https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/debian-9.3.0-amd64-netinst.iso

- **Do you see any meaningful error information in the DevTools?** 

No

lurch, 2018-02-24:
Etcher isn't able to burn CDs or DVDs - those need special CD-writer software. Etcher only writes to devices that appear to the OS as a standard disk-device (i.e. hard drives. USB flash drives, SD cards in a SD card-reader, etc.)
@jhermsmeier Could `drivelist` be modified to not include DVD-burners in its list of available drives?

rinchen, 2018-02-24:
That would explain it! Thanks.

jhermsmeier, 2018-03-20:
@lurch there's https://github.com/resin-io-modules/drivelist/issues/162, which is about the opposite – we could either add support for listing optical drives / media and add a flag to filter them easily, or stick to block devices only. But we should filter out optical media in Etcher either way

closed 2018-10-15 by jviotti