balena-io/etcher #2087: CLI download links 404

fenhl, 2018-02-25:
The Windows and macOS download links on https://etcher.io/cli/ point to [these](https://github.com/resin-io/etcher/releases/download/v1.3.1/Etcher-cli-1.3.1-win32-x64.zip) [three](https://github.com/resin-io/etcher/releases/download/v1.3.1/Etcher-cli-1.3.1-win32-x86.zip) [pages](https://github.com/resin-io/etcher/releases/download/v1.3.1/Etcher-cli-1.3.1-darwin-x64.tar.gz), all of which show GitHub 404 errors.

lurch, 2018-02-25:
Thanks for letting us know! This is related to #2072 
Looks like @jhermsmeier re-uploaded the Linux CLI archives to https://github.com/resin-io/etcher/releases/tag/v1.3.1 but forgot to also re-upload the Windows and Mac CLI archives. I'm sure this is something he'll be fixing soon... ;-)

lurch, 2018-02-26:
All sorted out :smiley:

closed 2018-02-26 by lurch