balena-io/etcher #2088: Keyserver Issue

Nastras, 2018-02-26:
- **latest** 
- **Debian Stretch Desktop Raspberry Pi:** 

Hello everybody,

I have etcher installed and get this error message when installing sudo get update:

`pi@HomeKitServer:~ $ sudo apt update
Ign:1 http://ftp.debian.org/debian stretch InRelease
OK:2 http://ftp.debian.org/debian stretch-updates InRelease                                                             
OK:3 http://archive.raspberrypi.org/debian stretch InRelease                                                            
OK:4 http://security.debian.org stretch/updates InRelease                    
OK:5 http://ftp.debian.org/debian stretch Release                            
OK:6 https://deb.nodesource.com/node_9.x stretch InRelease
Ign:7 https://dl.bintray.com/resin-io/debian stable InRelease
Holen:8 https://dl.bintray.com/resin-io/debian stable Release [1.878 B]
Holen:9 https://dl.bintray.com/resin-io/debian stable Release.gpg [821 B]
Ign:9 https://dl.bintray.com/resin-io/debian stable Release.gpg
Paketlisten werden gelesen... Fertig
W: GPG-Fehler: https://dl.bintray.com/resin-io/debian stable Release: Die folgenden Signaturen konnten nicht überprüft werden, weil ihr öffentlicher Schlüssel nicht verfügbar ist: NO_PUBKEY 379CE192D401AB61
E: The repository 'https://dl.bintray.com/resin-io/debian stable Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

If I do this here, I get this error:
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61

`pi@HomeKitServer:~ $ sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.X7aP7HFmpN/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: failed to start the dirmngr '/usr/bin/dirmngr': Datei oder Verzeichnis nicht gefunden
gpg: connecting dirmngr at '/run/user/0/gnupg/d.s65wqjo3nk9rke85qetdsjek/S.dirmngr' failed: Datei oder Verzeichnis nicht gefunden
gpg: keyserver receive failed: Kein Dirmngr
`
Does one have an idea where the problem is?

Thanks!

Nastras, 2018-02-26:
I have dirmngr installed now I get this error:

`pi@HomeKitServer:~ $ sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.lOvn8U940V/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: Keine Daten`

lurch, 2018-02-26:
Unfortunately it seems like the pgp.mit.edu keyserver occasionally has downtime issues. You can either try again later to see if it's back up, or https://github.com/resin-io/etcher/issues/1930#issuecomment-352845004 suggests an alternative might be to do:
```
curl https://bintray.com/user/downloadSubjectPublicKey?username=bintray | sudo gpg --dearmor --output /etc/apt/trusted.gpg.d/bintray.gpg
```
instead of using `apt-key`, although I've not tried this myself.

Nastras, 2018-02-26:
Hi, thank you very much for your reply. I tried it again earlier and it worked.

closed 2018-02-26 by Nastras

Querynalhabsyi, 2018-05-25:
@lurch Thanks, bro!

lurch, 2018-05-25:
See also #2319