balena-io/etcher #2092: Error: Unmount failed

Brnrr, 2018-02-27:
- **1.3.1:** 
- **Win 10 Pro 1709:** 
- **Zorin-OS-12.2-Core-64:** 
- **C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-
C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Tue Feb 27 2018 10:15:56 GMT-0800 (Pacific Standard Time) Flash error ({"error":{"message":"Unmount failed","stack":"Error: Unmount failed"},"image":"C:\\Users\\user\\Downloads\\Zorin-OS-12.2-Core-64.iso","drive":{"size":7948206080,"isVirtual":false,"enumerator":"SCSI","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"adapter":"standard","isSCSI":true,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"1.0","isSystem":false,"busType":"SD","isCard":true,"isUSB":false,"mountpoints":[{"path":"E:\\"}],"description":"Ricoh SD Disk Device","isUAS":false},"uuid":"baf45464-036c-499b-8b39-8cef818621c2","unmountOnSuccess":true,"validateWriteOnSuccess":true})
C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Tue Feb 27 2018 10:15:56 GMT-0800 (Pacific Standard Time) Open modal ({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 {"stack":"Error: Unmount failed","message":"Unmount failed"}
console.(anonymous function) @ C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55
C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 sdk start +6m true
C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Flash results Object
C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Terminating IPC server
C:\Users\user\AppData\Local\Programs\etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:55 Tue Feb 27 2018 10:16:04 GMT-0800 (Pacific Standard Time) Modal accepted ({"value":false})**

Brnrr, 2018-02-27:
I have tried to flash this image several times now and every time it fails in the unmounting process and then the SD card (8gb) become corrupted and needs reformatting. This most recent time the SD card is still readable after unmounting failed but now only reads as 24mb.

lurch, 2018-02-27:
Answering just the last part of your problem:

> This most recent time the SD card is still readable after unmounting failed but now only reads as 24mb.

That's totally normal behaviour, and is a side-effect of Windows not being able to read the filesystems used by Linux distros. You should find that your USB stick works fine if you actually try booting from it though?

jhermsmeier, 2018-03-09:
Duplicate of #1370

closed 2018-03-09 by jhermsmeier