balena-io/etcher #2094: Etcher refused to write to SD card

hedss, 2018-03-01:
- **Etcher version: 1.3.1** 
- **Operating system and architecture: OSX/El Capitan (10.11.6)** 
- **Image flashed: Raspberrypi3 from Devenv** 
- **Do you see any meaningful error information in the DevTools?: Sure do!** 

The normal modal dialogue asking for my user password did not appear prior to the flash attempt, which I assume is the reason for the problem.

Interestingly, restarting Etcher *did* work, but I was again not asked for my password. Curious.

Transcript below.

```
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:22 GMT+0000 (GMT) Reselect image ({"previousImage":{"path":"/Users/heds/Downloads/resindev-logpi-2.9.6+multi1.prod-2.9.6+multi1.prod.img.zip","size":{"original":1807745024,"final":{"estimation":false,"value":1807745024}},"extension":"img","archiveExtension":"zip","hasMBR":true,"hasGPT":false,"partitions":[{"type":14,"id":null,"name":null,"firstLBA":8192,"lastLBA":90112,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":90112,"lastLBA":729088,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":729088,"lastLBA":1368064,"extended":false},{"type":15,"id":null,"name":null,"firstLBA":1368064,"lastLBA":3530752,"extended":true}]}})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:22 GMT+0000 (GMT) Open image selector
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:27 GMT+0000 (GMT) Select image ({"path":"/Users/heds/Downloads/resindev-logpi-2.9.6+multi1.prod-2.9.6+multi1.prod.img.zip","size":{"original":1807745024,"final":{"estimation":false,"value":1807745024}},"extension":"img","archiveExtension":"zip","hasMBR":true,"hasGPT":false,"partitions":[{"type":14,"id":null,"name":null,"firstLBA":8192,"lastLBA":90112,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":90112,"lastLBA":729088,"extended":false},{"type":131,"id":null,"name":null,"firstLBA":729088,"lastLBA":1368064,"extended":false},{"type":15,"id":null,"name":null,"firstLBA":1368064,"lastLBA":3530752,"extended":true}],"logo":false,"bmap":false})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk stop +41s true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:36 GMT+0000 (GMT) Progress (undefined): 0% at 0 MB/s (eta undefineds)
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:36 GMT+0000 (GMT) Flash ({"image":"/Users/heds/Downloads/resindev-logpi-2.9.6+multi1.prod-2.9.6+multi1.prod.img.zip","drive":{"size":15931539456,"isVirtual":null,"enumerator":"diskutil","logicalBlockSize":null,"raw":"/dev/rdisk3","error":null,"isReadOnly":true,"displayName":"/dev/disk3","blockSize":null,"adapter":"standard","isSCSI":null,"isRemovable":null,"device":"/dev/disk3","busVersion":"0.0","isSystem":false,"busType":"UNKNOWN","isCard":null,"isUSB":null,"mountpoints":[],"description":"SD Card Reader","isUAS":null},"uuid":"74cae285-f088-48f9-86c2-f38854edba35","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Elevating command: /Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper /Applications/Etcher.app/Contents/Resources/app.asar
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Successfully connected to IPC server: etcher-server-82245, socket root /var/folders/jn/lxz1g0s173x99qb673tjh1qc0000gn/T/
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Image: /Users/heds/Downloads/resindev-logpi-2.9.6+multi1.prod-2.9.6+multi1.prod.img.zip
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Device: /dev/disk3
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Umount on success: true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Validate on success: true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:38 GMT+0000 (GMT) Flash error ({"error":{"message":"EACCES: permission denied, open '/dev/rdisk3'","stack":"Error: EACCES: permission denied, open '/dev/rdisk3'","code":"EACCES"},"image":"/Users/heds/Downloads/resindev-logpi-2.9.6+multi1.prod-2.9.6+multi1.prod.img.zip","drive":{"size":15931539456,"isVirtual":null,"enumerator":"diskutil","logicalBlockSize":null,"raw":"/dev/rdisk3","error":null,"isReadOnly":true,"displayName":"/dev/disk3","blockSize":null,"adapter":"standard","isSCSI":null,"isRemovable":null,"device":"/dev/disk3","busVersion":"0.0","isSystem":false,"busType":"UNKNOWN","isCard":null,"isUSB":null,"mountpoints":[],"description":"SD Card Reader","isUAS":null},"uuid":"74cae285-f088-48f9-86c2-f38854edba35","unmountOnSuccess":true,"validateWriteOnSuccess":true})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:38 GMT+0000 (GMT) Open modal ({"template":"./components/warning-modal/templates/warning-modal.tpl.html"})
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 {"stack":"Error: EACCES: permission denied, open '/dev/rdisk3'","message":"EACCES: permission denied, open '/dev/rdisk3'","code":"EACCES"}
console.(anonymous function) @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55
logException @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/resin-corvus/src/resin-corvus.js:228
exports.report @ /Applications/Etcher.app/Contents/Resources/app.asar/lib/gui/modules/exception-reporter.js:39
imageWriter.flash.then.catch @ /Applications/Etcher.app/Contents/Resources/app.asar/lib/gui/pages/main/controllers/flash.js:111
tryCatcher @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16
Promise._settlePromiseFromHandler @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504
Promise._settlePromise @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561
Promise._settlePromise0 @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606
Promise._settlePromises @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:681
Async._drainQueue @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:138
Async._drainQueues @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:148
Async.drainQueues @ /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:17
_tickCallback @ internal/process/next_tick.js:109
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk start +3s true
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Terminating IPC server
/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:55 Thu Mar 01 2018 11:50:42 GMT+0000 (GMT) Modal accepted ({"value":false})
```

jhermsmeier, 2018-03-01:
@hedss just to exclude the possibilities, can you confirm that:

- if using an SD-card, the physical lock-switch is not set?
- the password prompt is not hidden behind the Etcher window?

hedss, 2018-03-01:
@jhermsmeier I didn't touch the card at all between runs when it didn't, then did, write (and visual inspection showed the lock switch was not on).

The password prompt was not hidden behind the Etcher window. On restarting, the new instance wrote without prompting, and only at the next write did it ask for access.

jhermsmeier, 2018-03-01:
Thanks @hedss – that is odd, will see if I can reproduce somehow.

closed 2018-10-15 by jviotti