balena-io/etcher #2102: Etcher broke my USB

Octopous, 2018-03-03:
I've been flashing Ubuntu and Pop!_OS images to my Sandisk 64GB stick the past few days. Everytime it finished I was left with an unreadable drive by windows. No problem, since the stick boots fine and I assume windows can't read it because it's formatted as ext4. I had some trouble everytime to get the stick formatted in Windows again, I had to go into diskpart and "clean" the drive two times before it was formattable by windows again. This worked everytime.

However, after the last time I wrote the same image to the drive with Etcher, things changed. When I want to format it in windows it just won't work. The HP USB Disk Storage Format Tool, usually my lord and saviour, now tells me the USB is write protected. 

Things I tried:
- every possible method in diskpart
- regedit tweak to prevent drive write protection
- AOMEI Partition utility (funnily enough this actually seems to format the drive fine but in Disk Management it shows as RAW)
- Disk Management

How is a program like Etcher supposed to completely kill a USB drive? That's not supposed to happen am I right? The USB drive was working like a charm before the last deadly write by Etcher.

jhermsmeier, 2018-03-04:
Which Windows version, and which Etcher version are you running?

Octopous, 2018-03-04:
Newest Windows, newest Etcher at the time of writing. I fixed it by buying a new USB, booting a live ubuntu and formatting the "broken" drive with gparted. Still, this is a pretty big issue since not everyone has access to a linux environment and it seems to be unfixable in Windows.

lurch, 2018-03-11:
@flo10000 Have you looked at https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives ?

flo10000, 2018-03-11:
yes, thanks, but it doesn't work with those instructions . (I followed exactly how they say...) ... Do you know another way ?  thanks in advanced

flo10000, 2018-03-11:
Good news!!!  I did it in Terminal (with MAC OS)  writing the following command:

diskutil eraseDisk FAT32 DiskNameGoesHere /dev/DiskNodeIDHere

:)

jhermsmeier, 2018-03-20:
Good to hear :) Closing this then

closed 2018-03-20 by jhermsmeier

Nathanael8, 2018-06-11:
I am running a mac and have a similar problem, but the stated fix doesn't work.

jhermsmeier, 2018-06-11:
@Nathanael8 for the `eraseDisk` command you'll have to use the entire device, i.e. `/dev/disk2` instead of `/dev/disk2s2` (which refers to a partition on the device)

mchsk, 2018-06-17:
so is there a way to flash eg ubuntu iso with etcher on macos now? cause I broke two flash drives with it :D

jhermsmeier, 2018-07-04:
@hitchhooker you might want to try the latest Etcher version (1.4.4)

hitchhooker, 2018-07-04:
good to know, ill try some another software. can not take more risks for losing my mental health with writing software on windows environment. i have now been failing to install AntergOS on my pc for 4 months realizing it was all the time about the failure of software I was making the installation media.

lurch, 2018-07-04:
In my experience, Etcher has been pretty reliable. Are you sure it's not your USB sticks (or Windows permissions) that are faulty? :man_shrugging:  But see https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable and https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos in case that helps...?

hitchhooker, 2018-07-04:
2 last ones were straight from the shop. Other I manage to burn ISO couple times with etcher but failed installing, especially AntergOS(cnnchi crashed in installation). but after couple reflashes with etcher it become unusable. last usb2.0 16gb stick went straight unusable on the first try even without process of burning files beginning giving the error message, so that might have been also poor quality of memory stick.

mchsk, 2018-07-05:
the flash drives are physically ok, i can recover them even on macos (the os that is pretty unwilling to talk to anything else except from what it wants)

cant it be that the etcher initally patches the drive with GPT/GUID instead of MBR? cause I believe it was the case after fu*ked up flashing :)

lurch, 2018-07-05:
> cant it be that the etcher initally patches the drive with GPT/GUID instead of MBR?

Etcher just raw-writes whatever's in your image file directly to your destination drive - in that regard it's similar to the way `dd` operates. So if your image file contains an MBR your flash drive will have an MBR, but if your image file contains a GPT your flash drive will have a GPT.

mchsk, 2018-07-05:
if that is the case you would be simply using `dd` on linux. so I thing there is some other logic. when I use unetbootin, it does MBR I believe, or? Again, latest etcher on macos created gpt.

lurch, 2018-07-05:
> if that is the case you would be simply using dd on linux

:roll_eyes: 
https://resin.io/blog/etcher-1-0-is-here/
https://resin.io/blog/etcher-now-with-multi-write-and-compute-module-support/

czfries, 2018-07-11:
I thought I had broken a USB drive this way as well. However, if you go to Disk Utility and set the view option (top left corner) to Show All Devices, you can see the drive itself rather than the partition inside. Simply click on the drive and the click Erase. You can format the USB drive however you want to and it should work as it used to.

jhermsmeier, 2018-07-12:
Oh, thanks @czfries! This is actually a not unimportant detail – we should add this to our recovery guide docs.

k1ll3r8e, 2018-10-06:
Hey there,
had today the same issue after "Etching" my Sandisk Cruzer 8 GB.

Here is a simple solution for those using windows:
Press WIN/Logo+R type: **diskpart**
Now you should see the console of diskpart (looks like default cmd)
just type: **list disk** the ouput should look like:
`

    Datenträger XXX  Status         Größe    Frei     Dyn  GPT
     
    Datenträger 0    Online         1863 GB  1024 KB
    Datenträger 1    Online         1863 GB      0 B
    Datenträger 2    Online          465 GB      0 B
    Datenträger 3    Online          335 GB      0 B
    Datenträger 4    Online         7691 MB      0 B
`
in my case its the **7691 MB** device. Type: **select disk 4** (replace the ID with your device ID)
To make sure that you really got the right device, check it with **detail disk**. If the device match to your USB device, type: **clean** (This will **erase all data** on the device!)

The main part is done, now you can create a new partition with:
`
    
    create partition primary
    
    select partition=1
    
    format fs=fat32 quick
`
(You can also replace the fs=**fat32** with **ntfs**)

additional, you can set it **active** by just typing: **active**. Now you should be able to use your device by typing: **assign**

And magically a window pop's up which should ask you what you want do with your device ;)

Regards,
Sebastian

Wotzhs, 2018-12-02:
Here's how i fixed mine on windows 10, after using etcher to flash pop!_os iso to my usb drive:

`diskpart.exe`'s `clean` failed to complete and noticed the usb drive doesn't have gpt:
[image]

so i convert it to gpt and then run again and it completes:
[image]

then just do the regular format in windows disk management, and voila, usb drive can be used again.

thundron, 2019-01-18:
@cata0309 Chances are that you flashed an OS that yours cannot read fully, so it tries to read w/e which happens to be only 2MB.
What image did you flash on your USB?

cata0309, 2019-01-18:
Ubuntu iso

thundron, 2019-01-18:
@cata0309 That's it, your OS (the one you're running Etcher on) can't read that image's partitions correctly so it might as well give you a warning about it, which is totally fine, I receive warnings all the time on OSX when flashing Linux distros on sdcards/USBs.
Afaik ubuntu iso's should be bootable even if flashed by Etcher, you can verify it and let us know how it goes. If it doesn't boot though, you might need to use a different tool, check [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md) for more infos.

andreiconoval, 2019-01-24:
For restore USB on windows use EaseUS Partition Master 13.0 and Disk management.
1) Delete volume with Disk management and after create them , will be shown info that usb is not formatted.
2) Use  EaseUS Partition Master 13.0  to format usb  FAT32

techsolveprac, 2019-03-01:
Your flash drive can be recovered, read here --> https://www.techsolveprac.com/repair-flash-drive/

MarcelloLins, 2019-04-25:
Thanks @Wotzhs for the tip :)
Saved my ass

annymosse, 2019-05-15:
sorry to the about that but really FU## to your stupid software ,and ETch my :rocket: ,
your stupid 'Safe usage' broke my laptop and my USB too.

techsolveprac, 2019-05-16:
The USB may not be broken, the flash drive becomes unusable on Windows Operating System after flashing with Balena Etcher software as Linux File System is not readable by Windows. It may appear that the flash drive is no more appropriate for use. Still, the flash drive may be safe and can be repaired. Trying deleting the newly created partition in the type of OS you have flashed for and it may become reusable again. Read the complete [Article ](https://www.techsolveprac.com/repair-flash-drive/) or watch out the [Video](https://youtu.be/MoZG72vn-WQ).
Hope it is found useful.

utk-dev, 2019-06-20:
Etcher still breaks USB Drives. Do not use Etcher! I tried to make a pop os bootable drive and it messed my pen drive. No Windows trickery could resolve it. Ultimately, I used rufus and it worked.

USE RUFUS. THIS ETCHER THING IS A BIG SCAM.

simonjoeca, 2019-07-15:
test

kylegnally, 2019-07-20:
So I just managed to use Etcher to burn a Ubuntu 18.04 iso because the Universal USB Installer software did not properly install ubiquity and I thus could not actually install Ubuntu (I could only run the live USB).

Etcher did copy the iso properly,but reduced my USB stick from 64GB to 8GB in the process. 

ETCHER WILL REDUCE THE CAPACITY OF YOUR USB STICK. DO NOT USE THIS SOFTWARE.

For the record, I was able to use gparted to restore the correct size and format the stick as NTFS but lost the live USB in the process (of course). 

I will not be using this software again in the future and I will instruct other not to do so. This is BAD software and should be removed until it is fixed; a SO thread involving the ubiquity problem mentioned above tells users to use this and unfortunately I do not have enough "reputation" to make this suggestion (yet another reason I steer clear of that site).

Again:

DO NOT USE THIS SOFTWARE. 

ETCHER WILL BREAK YOUR MEDIA.

deadair3210, 2019-07-21:
NO IT WONT

WHY ARE YOU SCREAMING

It's surprising how many people don't realise how hard it actually is to break a flash drive other than just writing to it until it wears out. 

The reason that it shows up as a lower amount is the image you wrote was only that big. Any extra space is ignored until the drive is clean formatted which removes the partition table and restores the drive to the normal size

nhoj01, 2019-08-06:
Broke my flash drive on 8/5

twas on a windows 10 1809 device trying flash to tails. 3.15

techsolveprac, 2019-08-06:
https://github.com/balena-io/etcher/issues/2102#issuecomment-492973446

katktv, 2019-08-10:
For anyone coming by, Etcher writes the OS in a specific way so it is properly readable in Linux but not in Windows. If you, like me, need to recover the USB drive while on Windows, diskpart helped me as per this video.

https://www.youtube.com/watch?v=LfCjFY6YF68

techsolveprac, 2019-08-10:
Check this comment above. The only issue is Windows can't read Linux partition. So flash drive becomes unreadable on Windows. Check out here https://github.com/balena-io/etcher/issues/2102#issuecomment-492973446

newearthmartin, 2019-08-25:
Hello, trying to recover a disk broken on mac while writing a windows ISO with Etcher

The disk shows up in `diskutil list`:

```
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *253.8 GB   disk2
   1:               Windows_NTFS ELMARTUS                253.8 GB   disk2s1

```

But when I try `eraseDisk` it says: `Error: -69877: Couldn't open device`

```
$ diskutil eraseDisk Fat32 MYDISK disk2
Started erase on disk2
Unmounting disk
Error: -69877: Couldn't open device
```
Trying to write another iso fails also. Anyone could get over this?

Modi34, 2019-12-27:
broke my new usb drive on mac following this https://tutorials.ubuntu.com/tutorial/tutorial-create-a-usb-stick-on-macos#3

was able to restore it with 
`sudo fdisk -i /dev/disk7`

turtleisaac, 2019-12-27:
> broke my new usb drive on mac following this https://tutorials.ubuntu.com/tutorial/tutorial-create-a-usb-stick-on-macos#3
> 
> was able to restore it with
> `sudo fdisk -i /dev/disk7`

Any idea which version of Etcher and macOS you are using? I followed this same tutorial a while back to create an Ubuntu install USB and it worked flawlessly. I've used it for CentOS 7 as well and received the same results.

Ik these versions are now old, but when I made the install media I had macOS Mojave 10.14.4 and Etcher 1.4.4

Modi34, 2019-12-27:
I am using mac os 10.15.2 with etcher 1.5.70

I think something wrong with my ubuntu iso file 
I took it from http://cdimage.ubuntu.com/releases/16.04.6/release/ (Mac (PowerPC) and IBM-PPC (POWER5) server install image)

Tried creating usb with rufus on windows 10 and had same issue

Im not very familiar with linux so I probably screwup somewhere :)

deadair3210, 2019-12-27:
Etcher didn't break your drive, in fact it seems to have done exactly what you told it to do. It applied a PowerPC based Ubuntu iso to your drive. The reason it wouldn't show up in Finder is likely that APT. APT stands for Apple Partition Table, it's what Macs used to require to boot before they moved over to Intel. What are you trying to do and I might be able to help

marcus-sa, 2020-01-12:
Yeah, Etcher really broke my SanDisk drive. It's not even available when using `diskpart`

deadair3210, 2020-01-12:
Unless there is a bug in the firmware for your drive, there is literally no way for Etcher or any program to kill your drive. The only way for a program to kill your drive is to exhaust all the write cycles left in the flash memory

newearthmartin, 2020-01-12:
Can you "literally" read what people are saying?

Echter broke my drive, I couldn't repair it with windows, Linux or Mac. No
matter if the hardware is ok, all OSes seem to go crazy with it.

Modi34, 2020-01-12:
not sure if this tip is helpful for anyone but..
I solved all my problems by installing everything I need on my hard drive via parallels desktop
You simply mount your iso, dmg, img or whatever in one click, mount your ssd via usb cord and install everything you need...
after that I just put my ssd into my old macbook and it just works..... it took me less than 15 min to try what I wanted...

newearthmartin, 2020-01-12:
@jhermsmeier can you reopen this case and not as "support"? There is enough evidence with all these reports that etcher is breaking people's drives. 

If you don't do anything about it, then you are being negligent.

jhermsmeier, 2020-01-14:
@newearthmartin I stopped working on Etcher in late 2018, and am not with balena anymore. Also, the evidence (plenty of it in this and other threads) kind of tells the opposite story.
Anyways, you can always just open a new issue, if there's anything that didn't work for you.

newearthmartin, 2020-01-14:
I'm not getting the bad karma for negligence by ignoring all the people
that say that etcher broke their pen drives, and the future guys whose
pendrive you will break... So you do whatever you want, it's on you guys.
Our responsibility is to come here and report it. You are dropping the ball.

And what's the opposite story from evidence? That etcher didn't break
plenty of other pendrives. It should break none!

deadair3210, 2020-01-14:
There is no more evidence that Etcher breaks flash drives any more than Rufus, unetbootin, or any other image program does. Flash memory dies, it's not a if, it's a when. It's not like there is some evil command you can send to flash drives to kill them

newearthmartin, 2020-01-14:
It's not that my drive died coincidentally. Can you see how much you refuse
to take what we are saying?

I tried to Flash a windows ISO and etcher gave some warning and I
continued, and it broke my drive. So given that it gave me a warning,
etcher was aware that it might do something wrong.

Then I used unetbootin and flashed the same ISO in another drive without
problem. So there must be room to improve.

And apparently there is a state in which you can leave the drive in which
OSes fail to open the drive or fix it. If I knew how to make a dump of the
drive and send it, maybe you could use that for debugging. It's a 256gb
drive though

lurch, 2020-01-16:
> I tried to Flash a windows ISO and etcher gave some warning and I
continued, and it broke my drive. So given that it gave me a warning,
etcher was aware that it might do something wrong.

Do you remember what the warning was? If it was a Windows ISO, it was probably the warning about Windows ISOs not being bootable (i.e. nothing to do with "breaking drives").
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable
https://github.com/balena-io/etcher/blob/master/lib/shared/messages.js#L131

> It's a 256gb drive though

There's so much price-competition in the flash drive market, I suspect that the bigger the drive, the more chance there is that it's a "fake" flash drive.
https://sosfakeflash.wordpress.com/
https://fightflashfraud.wordpress.com/

PratyaySarkar, 2020-05-03:
I fixed my USB using rufus (if someone has not tried that method). It is the simplest way to do it.

deadair3210, 2020-05-03:
Most iso files are not bootable by default, you have to make them bootable. Etcher doesn't apparently do this while Rufus does.

lurch, 2020-05-04:
> Most iso files are not bootable by default, you have to make them bootable. Etcher doesn't apparently do this while Rufus does.

Indeed - in order to do that, Rufus has special knowledge about how to modify the image files (see #413 for more info). In contrast, Etcher doesn't modify the image and just writes it byte-by-byte onto the target drive. ISOs written by Etcher **are** bootable if they're in a [specific format](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos).

newearthmartin, 2020-05-04:
Etcher somehow knows that the iso is not bootable because it shows a
warning. It showed me a warning and I thought "what's the worst that could
happen?". Now my drive is unrecoverable. Etcher shouldn't write those isos
at all because it can break drives.

newearthmartin, 2020-05-04:
I tried all these methods in all os and no success

deadair3210, 2020-05-04:
1. It's up to the user to make decisions about what to do with their hardware/software. If the user wants to write literal random garbage, it's not the systems job to stop them. 

2. For hopefully the last time, it is literally impossible for a user level program to do anything to kill a USB drive, as the only failure states that would stop a drive from working is either a firmware bug or the flash memory dying

newearthmartin, 2020-05-04:
Evidence says otherwise. It's actually pointless to tell it to you guys,
your developer ego is so fragile that you won't take hundreds of reports
and always find a way to blame it on something else. Keep up the good work
breaking drives 👍🏻

newearthmartin, 2020-05-04:
It might as well be that the drives are not broken but the OSs get broken
when you put such drive, they don't know what to do and give up on the
drive.

[3 more replies]