balena-io/etcher #2107: Using the CLI

DovaMerith, 2018-03-07:
I can't seem to figure out how to list available drives and flash my img to it, can anyone help?
- **Etcher version:** 
1.3.1_CLI
- **Operating system and architecture:**
ChromeOS_32bit_Dev-with-Linux-shell 
- **Image flashed:** 
Need help

jhermsmeier, 2018-03-07:
You can call it without the `--drive` option to get an interactive list of drives;

```
$ sudo etcher path/to/image.img
? Select drive (Use arrow keys)
❯ /dev/disk2 (31.1 GB) - SD Card Reader
```

DovaMerith, 2018-03-07:
Thanks, i have done that, but now im getting one of two errors, "Invalid
Argument, Write" or "UNEXPECTED 15"

What do these errors mean? and how do i fix them?

lurch, 2018-03-08:
From what I can tell, I think the "Invalid Argument, Write" should be fixed by #2055 and thus will be fixed in the next release of the Etcher CLI (whenever that happens to be) ?
I've got no idea about your other error though.  @jhermsmeier Is there a "CLI equivalent" to the DevTools Console? i.e. some way of capturing any debug output (perhaps to a logfile).

jhermsmeier, 2018-03-21:
@DovaMerith The "UNEXPECTED 15" error is of the same nature as https://github.com/resin-io/etcher/issues/1697 – which has been fixed in the PRs #2095, #2097, and #2099 – while "Invalid Argument, Write" is most likely what @lurch mentioned, but either way, both of these should be fixed in the next release

jhermsmeier, 2018-03-21:
Closing as relevant PRs have already been merged.

closed 2018-03-21 by jhermsmeier