balena-io/etcher #2109: Adding dependency to files

aamm132, 2018-03-08:
I have been working on etcher locally for a bit, and I made it so it detects a specific image file on my machine and then when etcher is opened the file is selected already. However, if I run this version of etcher on a different machine the app will error endlessly. I think it has something to do with the fact that this image does not build correctly with the app or it is not detected somewhere. Is there a way I can change this, perhaps in the scripts it runs? Sorry if it's a bit too specific to me to be able to offer a solution, which is fine, but thanks for any help provided anyway!

jhermsmeier, 2018-03-08:
Alright, I have a couple questions to help me understand the situation better:

- Which operating system are you on?
- What build steps are you running?
- What are the errors it's spewing out?
- Between what kind of machines are you moving it? Same OS, same architecture (i.e. both 64bit)?
- Is the image supposed to be bundled with Etcher, or is it just in some location on those machines?

aamm132, 2018-03-09:
Whoops, sorry I was wrong about the error it spams. It actually spams there is no such file. I placed it in the assets folder in etcher, but it looks at the folder under C:/Users/USERNAME/AppData/Local/Temp/assets/

jhermsmeier, 2018-03-09:
The assets directory is used for electron-builder assets, meaning those don't end up in any filesystem related space when building Etcher. You'll want to put that somewhere else, a good place would be `lib/blobs/`.

aamm132, 2018-03-09:
I think that takes care of it thanks!

lurch, 2018-03-11:
Perhaps this indirectly links to #1386 ? i.e. being able to specify which image-file the Etcher GUI should open by default.

aamm132, 2018-03-21:
I have to unfortunately reopen this as I ran another build command causing the issues described above. If I run the portable installer (given through make electron-installer-portable) the build fails saying the image file I used cannot be found. I am guessing the file is not building correctly, so is there a NSIS file perhaps I need to change to include this file? Thanks again.

jhermsmeier, 2018-03-21:
@lurch #1386 is not about this, but about making Etcher understand default OS command line arguments, so that it'll open & use an image file if you drag it onto it, or start it as a command with an argument

jhermsmeier, 2018-03-21:
@aamm132 is it the build process itself that's failing, or the built Etcher, once you run it? Is the portable installer generate roughly the size of the image you're bundling?

aamm132, 2018-03-21:
Built etcher is failing once it is ran. It works fine doing the executable that is not the portable installed one. The portable installer is spamming there is no such file or directory, with regards to my image, when I run it.
My image file is about 1,440 KB while the portable installer generated is about 51,006 KB.

jhermsmeier, 2018-03-21:
Oh, that's a really tiny image. How are you resolving the path to it?

lurch, 2018-03-22:
> @lurch #1386 is not about this, but about making Etcher understand default OS command line arguments, so that it'll open & use an image file if you drag it onto it, or start it as a command with an argument

Correct, I guess I should have clarified my thinking a bit better... I meant that _if_ #1386 was implemented, which allowed you to start the Etcher GUI with an image-file on the command line, then @aamm132 would be able to solve his problem by distributing a zipfile containing Etcher, his image-file, and a batch-script, with the batch-script then doing something like
```
Etcher-Portable-1.3.1-x64.exe FILENAME.img
```
and he wouldn't need to try building a custom version of Etcher.

jhermsmeier, 2018-06-05:
@aamm132 sorry for the belated response; the issue with the path is likely that the html tpl ends up in the webpack bundle, which sits in the `generated/` directory, so the path will need adjusting to resolve from there I believe.

closed 2018-10-15 by jviotti