balena-io/etcher #2122: No such file or directory: undefined with version etcher-1.3.1-x86_64

dannn-o, 2018-03-14:
- Version: etcher-1.3.1-x86_64 
- OS: Linux Mint 17.3  
- Image: Fedora 27 Cinnamon 
- Do you see any meaningful error information in the DevTools?:  
```

{"stack":"Error: ENOENT: no such file or directory, open '/run/user/0/drivelist-f5b0a11d7fc6.sh'","message":"ENOENT: no such file or directory, open '/run/user/0/drivelist-f5b0a11d7fc6.sh'","code":"ENOENT"}
console.(anonymous function) @ /tmp/.mount_nutRHU/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:55
/tmp/.mount_nutRHU/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:55 sdk start +5m true
/tmp/.mount_nutRHU/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:55 Flash results Object
/tmp/.mount_nutRHU/usr/bin/resources/app.asar/node_modules/raven-js/dist/raven.js:55 Terminating IPC server
```

-----------------------------------------------------------------

I've been using etcher-1.0.0-linux-x64 on my Linux Mint 17.3 laptop for quite a while now without any issues.  I decided to get a bit more current and downloaded the newer etcher-1.3.1-x86_64 version.
Things seemed about the same, up until the last step (Flash!).  I was now prompted to enter the super user password (needed for /usr/bin/env.  "Details" shows action of org.freedesktop.policykit.exec).

After entering the password, an error window appears:
 ```
     No such file or directory: undefined
     The file you're trying to access doesn't exist
```
After clicking OK to clear the error, another Attention window displays:
 ```
      Oops, seems something went wrong (Retry).`
 ```
If I run the exact same process using version 1.0.0, I am not prompted for the password and the flash completes successfully.  Image is on an USB external hard drive, and is being written to a 16 GB USB 3.0 flash drive.

lurch, 2018-03-14:
@jhermsmeier It looks like the TMPDIR for the elevated (root) user `/run/user/0/` doesn't contain the drivelist script; presumably because it's only in the (different) TMPDIR for the normal user. At least, that's my best guess without attempting to reproduce this myself.
Similar-but-different to #1699 :wink: 

This might be fixed by https://github.com/resin-io-modules/drivelist/pull/255 ?

jhermsmeier, 2018-03-14:
Yeah, that's going to fix it, and all the other external script issues drivelist causes as well :)

closed 2018-10-15 by jviotti

dannn-o, 2018-10-16:
Is the fix available for me to test?

lurch, 2018-10-16:
There's been several releases of Etcher since 1.3.1 it seems. Perhaps a newer version fixes it?

dannn-o, 2018-10-17:
I just downloaded version 1.4.5 (latest), and tried to recreate the original scenario as closely as I could, though my main computer is now running Mint 18.3.  I am no longer receiving the fatal error when flashing the image ("No such file or directory: undefined") and it appears to complete successfully.

I still receive the prompt to "enter the super user password (needed for /usr/bin/env)", which I don't know if it's now a required action.  I can only say it had not been required in the previous version of etcher (under Mint 17.3).

jviotti, 2018-10-18:
@dannn-o 

> I still receive the prompt to "enter the super user password (needed for /usr/bin/env)", which I don't know if it's now a required action. I can only say it had not been required in the previous version of etcher (under Mint 17.3).

Etcher needs sudo privileges to write to removable drives. Some Linux setups allow users to do so without extra permissions (very distro/setup dependent) and thus Etcher might have worked for you before without even showing this dialog, but it has been present since the very beginnings.

davidmc5, 2018-10-18:
I just downloaded version 1.4.5 on Windows10 and still produces the error. The workaround that worked for me was to move the image file from the NAS repository to the windows desktop.

Garfonso, 2018-10-27:
I have the same issue as davidmc5. I made sure the super user has the netwrok drive mapped to the same letter and access to it. But it does not help.

lurch, 2018-11-27:
@davidmc5 and @Garfonso see #2278