balena-io/etcher #2127: Error Opening Etcher if Etcher's Obtrusive Internet Requests are Blocked by Firewall

ClosetCoder, 2018-03-20:
- **Etcher version:** 
     v1.3.1

- **Operating system and architecture:** 
     mac OS High Sierra v10.13.3 (17D102) - i7 Quad 2.4GHz - 16GB RAM - 512GB SSD
- **Image flashed:** 
     None - Error occurs simply opening app if internet access is blocked
- **Do you see any meaningful error information in the DevTools?** 
     Yes:  [Error-A](https://ibb.co/cQcNZc)  &  [Error-B](https://ibb.co/esy70H)
- **Error message produced by Etcher**
     Yes:  [Error Message](https://ibb.co/mTnuxx) w/'OK' to proceed
- **Clicking 'OK' on message allows Etcher to run**
     Yes
- **Define Blocked in Issue Title**
     Rules are added to firewall to block ALL internet requests made by Etcher

jhermsmeier, 2018-03-20:
Inlining images here for quicker reference;

Error A

[image]

Error B

[image]

Error Message

[image]

lurch, 2018-03-22:
Looks like this is closely related to #2126 ? (just with a slightly different error dialog at the last step)

BananaAcid, 2018-04-08:
And possibly #2143 (for Windows).

closed 2018-10-15 by jviotti