balena-io/etcher #2128: Documentation on how to use bmap files with etcher

symgryph, 2018-03-21:
Just a documentation question on bmaptool support, where do I put the bmap file and what do I name it to?

Thomas

jhermsmeier, 2018-03-21:
The documentation around that has been removed, but never really replaced with something.
All you need to do though, is to put both the image, bmap and a manifest into a zip archive. The manifest is just a JSON file in a `.meta` directory within the archive. The basic structure is the following:

```
archive.zip
├── imagefile.img
└── .meta/
    ├── manifest.json
    ├── image.bmap
    └── logo.svg
```

I'd recommend checking out https://github.com/ev3dev/ev3dev/releases for an example.

symgryph, 2018-03-22:
Thank you much for this. It really helps. I will be happy to submit a document with the entire process if you would like in markdown format. Should I do a pull request with the file?

Also, what method do you use to generate the bmap files? I use the bmaptool method, but its quite cumbersome, and only works on linux. Have you developed a more streamlined method of making bmap files?

Perhaps a docker appliance?

Thomas

jhermsmeier, 2018-03-22:
Sure, that'd be awesome if you'd PR it! We have a somewhat resinOS specific process to blockmap those images, which doesn't work with any image, and otherwise we rarely use bmaps to be honest :) The process they recommend is quite cumbersome, and also platform specific unfortunately, and we've been planning to create alternative methods, but it's not high on the list.

closed 2018-10-15 by jviotti

jcormier, 2021-03-29:
A 2019 example of a zipped image with bmap. https://github.com/balena-io-modules/etcher-sdk/commit/574978b8444eab0867d5ec4302d2c40ecac42707

jcormier, 2021-03-29:
bmap file needs to be named image.bmap and placed in the .meta folder

Then blockMap should be true in the console log after you select the image and it will print all the different blocks.

[image]

jcormier, 2021-03-29:
Example bash which renames wic image files from Yocto builds and zips them up

```bash
					# Rename wic to .img and zip up sd card images
					for f in *.wic; do
						# Replace .wic with .img
						img="$(basename -s .wic "$f").img"

						# Cleanup leftover .meta folders
						rm -rf .meta

						cp "$f" "$img"
						cp "$f".bmap "$img".bmap

						# Create .meta folder for balenaEtcher bmap support
						mkdir .meta
						cp "$f".bmap .meta/image.bmap

						md5sum "$img" > "$img.md5"
						"$ZIP" -r "$img".zip "$img"* .meta

						mv "$img".zip "${DEPLOY_DIR}/"

						# Cleanup .meta folders
						rm -rf .meta
					done
```